rpng/open_vins #9: how to run the system with four cameras?

YiChenCityU, 2019-08-20:


goldbattle, 2019-08-20:
Thanks for taking interest in the repository.

Right now only simulation supports the use of multiple cameras. If you want to use multiple cameras with real sensors, you will need to change feed functions in the [viomanager](https://github.com/rpng/open_vins/blob/5519764fb3f245ce6dd87fd15559b7e61075d859/ov_msckf/src/core/VioManager.h#L77-L93) and pass the images correctly from the main cpp files.

It is important to note that right now only synchronously triggered cameras are supported, thus given 4 camera images, perform tracking either mono or stereo (or any combination of the two) for your cameras. After that you can call your propagation/state cloning function and then update as normal.

If you have an example dataset that you are interested in running on, please post it!

YiChenCityU, 2019-08-20:
Thanks very much. This is a great open source project. Will you update part of the system based on this paper: Multi-Camera Visual-Inertial Navigation with Online Intrinsic and Extrinsic Calibration? I admire Professor Huang's work very much.

goldbattle, 2019-08-20:
This code was used as a based and edited for that paper, we are not planning on open sourcing that at this point, but maybe in the *far* future we would. This code is focused on readability and having a pretty minimal feature set which most people want. Thanks!

closed 2019-08-21 by goldbattle