rpng/open_vins #12: How to run simulation properly?

yuweijheng, 2019-09-03:
Hi, I am currently running the simulation by the command `roslaunch ov_msckf pgeneva_sim.launch` with the original provided launch file. However, I am not able to run the simulation properly. It shows that the average RMSE grows so fast and none of the features are used in MSCKF update. Any advice? Thanks!
[image]

goldbattle, 2019-09-03:
Thanks for checking out the project!
I am running this launch file on my local machine and it seems to be working. I know there was some changes at some point which changed how the first pose in the bspline was selected for initialization. What commit are you on? the latest? what does rviz look like?

For a seed of 3 I am getting my initial state to be different from the one you posted:
```
[ INFO] [1567514104.371065535]: [INIT]: INITIALIZED FROM GROUNDTRUTH FILE!!!!!
[ INFO] [1567514104.371105730]: [INIT]: orientation = 0.8005, -0.0274, -0.0477, 0.5968
[ INFO] [1567514104.371124984]: [INIT]: bias gyro = 0.0000, 0.0000, 0.0000
[ INFO] [1567514104.371140650]: [INIT]: velocity = 0.5283, -0.4576, 0.0844
[ INFO] [1567514104.371157069]: [INIT]: bias accel = 0.0000, 0.0000, 0.0000
[ INFO] [1567514104.371179438]: [INIT]: position = 0.1017, -0.0725, 0.0552
```

yuweijheng, 2019-09-04:
I am on the latest commit with Ubuntu 18.04. The estimated trajectory deviates from the ground truth in just a few seconds (position error after the starting system in 0.5 seconds is 1.397 meters). 
[image]

I then tried to run the default config with `euroc_V1_01_easy.txt`. I think that there is something wrong on my local machine with the initialization from ground truth. The reported initial state is
`[ INFO] [1567559600.933102781]: [INIT]: INITIALIZED FROM GROUNDTRUTH FILE!!!!!`
`[ INFO] [1567559600.933208277]: [INIT]: orientation = 0.7879, -0.0208, -0.0323, 0.6146`
`[ INFO] [1567559600.933242518]: [INIT]: bias gyro = 0.0000, 0.0000, 0.0000`
`[ INFO] [1567559600.933257819]: [INIT]: velocity = 0.6436, 0.2722, 1.2388`
`[ INFO] [1567559600.933270686]: [INIT]: bias accel = 0.0000, 0.0000, 0.0000`
`[ INFO] [1567559600.933289120]: [INIT]: position = 0.0281, 0.0191, 0.0869`
But the initial state should be something like
`[ INFO] [1567559600.933102781]: [INIT]: INITIALIZED FROM GROUNDTRUTH FILE!!!!!`
`[ INFO] [1567559600.933208277]: [INIT]: orientation = -0.8084, -0.0940, -0.5759, 0.0769`
`[ INFO] [1567559600.933242518]: [INIT]: bias gyro = 0.0000, 0.0000, 0.0000`
`[ INFO] [1567559600.933257819]: [INIT]: velocity = 0.1617, 0.0869, 0.3692`
`[ INFO] [1567559600.933270686]: [INIT]: bias accel = 0.0000, 0.0000, 0.0000`
`[ INFO] [1567559600.933289120]: [INIT]: position = 0.9295, 2.2087, 1.0252`

goldbattle, 2019-09-04:
Interesting, I am testing on 16.04.
Are you able to run it on real datasets?

yuweijheng, 2019-09-04:
Tested with EuRoC dataset and worked perfectly.

goldbattle, 2019-09-04:
Interesting. I know that the initialization from groundtruth is pretty sensitive on the real-world datasets, but I have never had any issues with the simulation. The initialized state shouldn't match anything in the file (but it should be close) since we use the poses in the file as the "control points" that we simulate our [b-spline](https://docs.openvins.com/simulation.html) over.

The simulator first tries to find a place in the b-spline where the IMU starts moving, and then initialize after this is detected. This mainly is to skip sections in the dataset where the unit is sitting still and monocular cameras will start to drift because they cannot triangulate features. This might be what is causing issues here. Can you try setting a smaller threshold here:
https://github.com/rpng/open_vins/blob/master/ov_core/src/sim/Simulator.cpp#L84

Can you also try simulating with two cameras?

yuweijheng, 2019-09-04:
Thanks for your illustrations. Actually, the initial state that I posted early is calculated by b-spline and close to the values provided by the file. I have tried a smaller threshold (0.01) and two cameras, but the results are pretty similar to the default config (estimated trajectory deviates from the ground truth in just a few seconds).

yuweijheng, 2019-09-05:
The calculated values of `pose0.inverse()`,  `pose1.inverse()` and `pose2.inverse()` in `BsplineSE3.cpp`  are somehow minus one times the correct values. After modifying these values in `BsplineSE3.cpp`, the simulator is now running perfectly.

goldbattle, 2019-09-05:
Interesting, sounds like maybe an Eigen issue, what version are you using?
Could you post what you did to address this so I can investigate a bit?

yuweijheng, 2019-09-05:
Eigen version: 3.3.4
Change `pose.inverse()` in
[https://github.com/rpng/open_vins/blob/master/ov_core/src/sim/BsplineSE3.cpp#L84](url)
[https://github.com/rpng/open_vins/blob/master/ov_core/src/sim/BsplineSE3.cpp#L124](url)
[https://github.com/rpng/open_vins/blob/master/ov_core/src/sim/BsplineSE3.cpp#L125](url)
[https://github.com/rpng/open_vins/blob/master/ov_core/src/sim/BsplineSE3.cpp#L126](url)
[https://github.com/rpng/open_vins/blob/master/ov_core/src/sim/BsplineSE3.cpp#L165](url)
[https://github.com/rpng/open_vins/blob/master/ov_core/src/sim/BsplineSE3.cpp#L166](url)
[https://github.com/rpng/open_vins/blob/master/ov_core/src/sim/BsplineSE3.cpp#L167](url)
[https://github.com/rpng/open_vins/blob/master/ov_core/src/sim/BsplineSE3.cpp#L224](url)
[https://github.com/rpng/open_vins/blob/master/ov_core/src/sim/BsplineSE3.cpp#L225](url)
[https://github.com/rpng/open_vins/blob/master/ov_core/src/sim/BsplineSE3.cpp#L226](url)
to `pose_inverse`, where
```
pose_inverse.block(0, 0, 3, 3) = pose.block(0, 0, 3, 3).transpose();
pose_inverse.block(0, 3, 3, 1) = -pose_inverse.block(0, 0, 3, 3) * pose.block(0, 3, 3, 1);
```

goldbattle, 2019-09-09:
Please let me know if commit https://github.com/rpng/open_vins/commit/23977bc0d7c2a204ee4dcd9594cb85ed2da8ed42 fixes the problem for you. Thanks!

yuweijheng, 2019-09-10:
It fixes the problem. Thanks!

closed 2019-09-10 by goldbattle