rpng/open_vins #13: Jacobian with respect to time offset

yuweijheng, 2019-09-10:
Hi, authors. Hope I don't get the code wrong, but it seems that the jacobian matrix with respect to time offset is not calculated in <https://github.com/rpng/open_vins/blob/master/ov_msckf/src/update/UpdaterHelper.cpp>?

kevineck, 2019-09-10:
Hi, we are using the cloning logic introduced in [Li and Mourikis](https://journals.sagepub.com/doi/pdf/10.1177/0278364913515286) (See equations 36-37). In particular, the clones are expressed at the true imaging time, and therefore the process of generating these clones already takes into account the time offset (see the [augment_clone](https://github.com/rpng/open_vins/blob/master/ov_msckf/src/state/StateHelper.cpp) function). Because the clones are already expressed at the true time, we do not need to add a measurement Jacobian for the time-offset.

goldbattle, 2019-09-10:
We just updated the documentation website with some more details.
Please take a look as see if that helps: [documentation link](https://docs.openvins.com/classov__msckf_1_1StateHelper.html#a598854d6e72b64a88e68fcc4a515bdb4)

yuweijheng, 2019-09-10:
Thanks for your illustrations. One last question, a bit confused why augment covariance for all rows and columns in 
<https://github.com/rpng/open_vins/blob/master/ov_msckf/src/state/StateHelper.cpp#L476> and
<https://github.com/rpng/open_vins/blob/master/ov_msckf/src/state/StateHelper.cpp#L477>, instead of the first `Cov.rows() - 6` rows and columns, i.e., 
`Cov.block(0, pose->id(), Cov.rows() - 6, 6) += Cov.block(0, state->calib_dt_CAMtoIMU()->id(), Cov.rows() - 6, 1) * dnc_dt.transpose();` and
`Cov.block(pose->id(), 0, 6, Cov.rows() - 6) += dnc_dt * Cov.block(state->calib_dt_CAMtoIMU()->id(), 0, 1, Cov.rows() - 6);`?

kevineck, 2019-09-10:
This can be understood in two ways. In way one, if you take the total Jacobian for the whole state, J, and write out JPJ^T blockwise, you will see required terms.

Alternatively, let us consider the linearized propagation of just the clone c,   c' = c+Nt, where c' is the error state of the clone at the true time, c is the error state at the estimated time, t is the time offset, and N is the time offset Jacobian.

To compute the marginal covariance of c' using the expectation, we have:

E[c' c'^T] = E[(c+Nt)(c+Nt)^T] =

E[c c^T]+NE[t c^T] + E[c t^T] N^T+E[t t^T] =

P_cc + N P_tc + P_ct N^T + N*P_tt*N^T.

In the solution you have written, the middle two entries would not exist,
and we would only have P_cc + N*P_tt*N^T, which ignores correlations.

In summary, because the entire state is correlated, we need to be careful when propagating/cloning.

yuweijheng, 2019-09-11:
Got it. Thanks!

closed 2019-09-11 by yuweijheng