rpng/open_vins #14: imu noise in Propagator.cpp

pzhu008, 2019-09-10:
Hi, 
As the default noise parameter  _noises is in the continuous case, for the bias driving noise,  should we use **_noises.sigma_wb_2*dt**  instead of **_noises.sigma_wb_2/dt** to convert to discrete according to (130)?

// Note that we need to convert our continuous time noises to discrete
// Equations (129) amd (130) of Trawny tech report
Eigen::Matrix<double,12,12> Qc = Eigen::Matrix<double,12,12>::Zero();
Qc.block(0,0,3,3) = _noises.sigma_w_2/dt*Eigen::Matrix<double,3,3>::Identity();
Qc.block(3,3,3,3) = _noises.sigma_a_2/dt*Eigen::Matrix<double,3,3>::Identity();
 Qc.block(6,6,3,3) = _noises.sigma_wb_2/dt*Eigen::Matrix<double,3,3>::Identity();
 Qc.block(9,9,3,3) = _noises.sigma_ab_2/dt*Eigen::Matrix<double,3,3>::Identity();

kevineck, 2019-09-10:
Hi,

Looking at  the noise jacobians, we have defined, for instance:
G.block(ba_id, 9, 3, 3) = dt*Eigen::Matrix<double,3,3>::Identity();

Thus we are defining b_a' = b_a + (n_ba_d) dt, where n_ba_d is the discrete noise, with covariance (1/dt)*sigma_ab_2 . Note that this dt will partially "cancel" the 1/dt sigma_ab_2 when computing Qd = G Qc G^T, and thus we will in total be left with the more familiar E[(b_a'-b_a)(b_a'+-ba)^T] = dt sigma_ab_2. The reason we wrote it this way was so that all the Qc would be more uniform. 

Please see [here](https://docs.openvins.com/propagation.html#error_prop) or details.

Therefore the "bias noise prop" that we define

pzhu008, 2019-09-10:
Hi Kevineck,

I see. Thanks.

pzhu008, 2019-09-10:
Hi,

Two more questions in the propagation part. 
(1) why to use **-w_hat * dt** (not w_hat * dt) to calculate exp_so3 and Jr_so3 ?
(2) when using the "summed" state transition and IMU noise to propagate the Cov as 
Qd_summed = F * Qd_summed * F.transpose() + Qdi;-------(1)
Cov.block(imu_id,imu_id,15,15) += Qd_summed; ------(2)

for axample: after two step propagation 
P_{k+2} = (F^2)*P_{k}*(F^2)^T+F*Q*F^T+Q;
while it seems that by using (1) and (2), we have:
P_{k+2} = P_{k}+F*Q*F^T+Q;

Thanks for your help~

kevineck, 2019-09-11:
Hi,
(1)-
This is because we follow the convention laid out in Trawny's tech report, wherein we parameterize the rotation from the global frame to the local frame R_GtoI rather than the (probably more common) rotation from the local frame to the global frame R_ItoG

In the other convention, we would have an update as  R_ItoG' = R_ItoG Exp(w dt), see for instance eq (23) in [Forster et al. ](http://www.roboticsproceedings.org/rss11/p06.pdf)

Because we use the transpose of that representation, we have R_IGoI' = Exp(-w dt) R_GtoI.

(2) In the line above (2) we take care of applying the state transition matrix to the covariance

Cov.block(imu_id,0,15,state->n_vars()) = Phi_summed*Cov.block(imu_id,0,15,state->n_vars());
Cov.block(0,imu_id,state->n_vars(),15) =Cov.block(0,imu_id,state>n_vars(),15)*Phi_summed.transpose();
    Cov.block(imu_id,imu_id,15,15) += Qd_summed;

If you write out the first two equations, you will see a Phi P_{k} Phi^T appear in the IMU's marginal covariance. After this, we then we add the Qd_summed term.

Also note that the "F" as you have written is the state transition due to a single IMU reading. Over this interval we have many IMU readings, so we compound them into a single state transition and noise term (which is eq (1) that you sent)

Hope this helps!

pzhu008, 2019-09-11:
Got it. Thanks, Kevineck.

closed 2019-09-11 by pzhu008