rpng/open_vins #15: questions about feature tracking

pzhu008, 2019-09-16:
Hi, 
I have two questions about the tracking part and really hope you could help.

(1) in the part of "check maximum baseline" in FeatureIntializer.cpp. Is there any reference about the way we calculate the base_line as
double base_line = ((Q.block(0,1,3,2)).transpose() * p_CiinA).norm(); 

(2)In the part of KLT based stereo tracking, only features that are successfully tracked in both cameras are good features. 
If we do not do feature matching between left and right cameras and only do KLT individually on each camera. with the extrinsics of two cameras, features that cannot be mathched between two cameras, but can be tracking on one of the cameras will also be used to update the state. Do you have any idea about the performance of this two methods? 

Thanks,

goldbattle, 2019-09-17:
(1) So Q will have the orthogonal directions to the pFinA, thus this is basically doing the dot product for each two orthogonal unit vectors with p_CiinA. More simply, this is projecting p_CiinA onto the plane whose normal is defined by pFinA. This distance norm is the "parallax" in this created plane. This is different from just taking the norm distance difference ||pCiinA|| as we are only interested in the parallax between frames, since moving towards a feature does not add any information.

(2) So if you don't enforce the stereo tracks you get the "duo" camera tracking setup. I recommend you checkout this paper by [Mrinal Paul](https://ieeexplore.ieee.org/abstract/document/7989022), which talks about what you mention. Basically, you don't gain any accuracy in your triangulation step and only gain in getting more measurements as compared to stereo. We also explored this in our multi-camera paper [here](https://ieeexplore.ieee.org/document/8793886).

pzhu008, 2019-09-17:
Hi Patrick,
It helps a lot. Thanks.

closed 2019-09-17 by pzhu008