rpng/open_vins #19: How to run open_vins with Live Camera?

arjunskumar, 2019-09-30:


goldbattle, 2019-09-30:
You can checkout the `run_subscribe_msckf` node which subscribes to ros topics.
Take a look at the [pgeneva_eth_ros.launch](https://github.com/rpng/open_vins/blob/master/ov_msckf/launch/pgeneva_eth_ros.launch) or [pgeneva_tum_ros.launch](https://github.com/rpng/open_vins/blob/master/ov_msckf/launch/pgeneva_tum_ros.launch) if you need an example.

arjunskumar, 2019-10-01:
@goldbattle Thanks. I made it working with realsense d435i, but one question does this algorithm requires imu excitation like in vins-mono? I observed a similar behaviour.

closed 2019-10-01 by arjunskumar

goldbattle, 2019-10-01:
If you are using a monocular system then yes you will need to excite the IMU to start the system.
There are a few thresholds in the launch files which can be tuned for your sensor / sensitivity.
```xml
<param name="init_window_time"       type="double" value="1.0" />
<param name="init_imu_thresh"        type="double" value="0.8" />
```
If you have stereo then you can set the threshold to zero so it initializes right away. This is possible since stereo allows for triangulation of features while standing still. The d451i color is rolling shutter? We don't have support for that so the accuracy will likely be low due to not modeling of it, but you could try using the IR cameras. We have had success using the new [t265](https://www.intelrealsense.com/tracking-camera-t265/) tracking camera.

hoangthien94, 2019-10-31:
Hi @goldbattle, would you be able to share the launch file/configuration that works for the T265 camera? I have been trying with different settings but could not get a good result so far.

hoangthien94, 2019-10-31:
I have it working now. It seems I used the wrong extrinsic params. 

For anyone else, you can replace this line in the [t265_stereo.py](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/t265_stereo.py#L157) script with the following to get the extrinsic params between the left fisheye camera and IMU (default is camera and `pose`, which is not the same as IMU, see [this diagram](https://github.com/IntelRealSense/librealsense/blob/master/doc/t265.md#calibration)):
```
get_extrinsics(streams["left"], profiles.get_stream(rs.stream.gyro))
```

robot-mh, 2020-01-23:
Hi @thien94 @arjunskumar , would you be able to share the launch file/configuration that works for the live camera? I have been trying with d435i with different settings, but also could not get a good result so far. 
I have calibrate the parametors well by kalibr, and run well by VINS-mono, but openvins can't be    running, it fly quickly.
@goldbattle

goldbattle, 2020-01-23:
You might need to set the `init_window_time` and `init_imu_thresh` to different values based on your IMU. I would recommend setting `init_imu_thresh` to very large, and then picking up the sensor from stationary to see what value would be a good threshold. We don't have structure from motion initialization like vins-mono, so you will need to start from stationary. This threshold is basically a check to see if there has been a large change in acceleration and thus, the filter should initialize.

Ador2, 2022-01-12:
Is this still working? links are dead, how to run on a live camera now?

goldbattle, 2022-01-12:
Simply run with the subscribe.launch file after you create your configuration.
https://github.com/rpng/open_vins/blob/master/ov_msckf/launch/subscribe.launch

The documentation has a tutorial:
https://docs.openvins.com/gs-tutorial.html