rpng/open_vins #21: What's the advantage of MSCKF features and SLAM featurs?

canyou2014, 2019-10-23:


goldbattle, 2019-10-23:
MSCKF features prevent the need for storing their positions in the state vector. This allows for more features to be used over the whole trajectory without increasing the computational burden by a lot. The temporal SLAM features are stored in the state vector until they are lost, and thus are good for features that are tracked for long periods of time (and thus provide loop closure information to limit drift). This paper might be of interest for you: http://www.roboticsproceedings.org/rss08/p31.pdf

closed 2019-10-28 by goldbattle

canyou2014, 2019-11-21:
@goldbattle  Thanks.  Another question: Will SLAM features converge more rapidly?

goldbattle, 2019-11-21:
Since SLAM features allow for measurements at future timesteps, yes they will be able to converge to better estimates. But at triangulation / initialization time they are equivalent in accuracy.