rpng/open_vins #23: Error whild building ov_core

bckim1318, 2019-10-29:
[build.make.004.log](https://github.com/rpng/open_vins/files/3781893/build.make.004.log)

I'm trying to build open vins, but it fails. How can I fix it?

bckim1318, 2019-10-29:
> `[ 68%] Built target ov_core_lib [ 73%] �[32m�[1mLinking CXX executable /home/ivy/workspace/catkin_ws_ov/devel/.private/ov_core/lib/ov_core/test_tracking�[0m [ 78%] �[32m�[1mLinking CXX executable /home/ivy/workspace/catkin_ws_ov/devel/.private/ov_core/lib/ov_core/test_repeat�[0m [ 84%] �[32m�[1mLinking CXX executable /home/ivy/workspace/catkin_ws_ov/devel/.private/ov_core/lib/ov_core/test_simulator�[0m /home/ivy/workspace/catkin_ws_ov/devel/.private/ov_core/lib/libov_core_lib.so: undefined reference to `cv::aruco::detectMarkers(cv::_InputArray const&, cv::Ptrcv::aruco::Dictionary const&, cv::_OutputArray const&, cv::_OutputArray const&, cv::Ptrcv::aruco::DetectorParameters const&, cv::_OutputArray const&, cv::_InputArray const&, cv::_InputArray const&)'
> collect2: error: ld returned 1 exit status
> /home/ivy/workspace/catkin_ws_ov/devel/.private/ov_core/lib/libov_core_lib.so: undefined reference to `cv::aruco::detectMarkers(cv::_InputArray const&, cv::Ptr<cv::aruco::Dictionary> const&, cv::_OutputArray const&, cv::_OutputArray const&, cv::Ptr<cv::aruco::DetectorParameters> const&, cv::_OutputArray const&, cv::_InputArray const&, cv::_InputArray const&)' CMakeFiles/test_repeat.dir/build.make:219: recipe for target '/home/ivy/workspace/catkin_ws_ov/devel/.private/ov_core/lib/ov_core/test_repeat' failed make[2]: *** [/home/ivy/workspace/catkin_ws_ov/devel/.private/ov_core/lib/ov_core/test_repeat] Error 1 collect2: error: ld returned 1 exit status CMakeFiles/Makefile2:2418: recipe for target 'CMakeFiles/test_repeat.dir/all' failed make[1]: *** [CMakeFiles/test_repeat.dir/all] Error 2 make[1]: *** Waiting for unfinished jobs.... CMakeFiles/test_simulator.dir/build.make:219: recipe for target '/home/ivy/workspace/catkin_ws_ov/devel/.private/ov_core/lib/ov_core/test_simulator' failed make[2]: *** [/home/ivy/workspace/catkin_ws_ov/devel/.private/ov_core/lib/ov_core/test_simulator] Error 1 CMakeFiles/Makefile2:2381: recipe for target 'CMakeFiles/test_simulator.dir/all' failed make[1]: *** [CMakeFiles/test_simulator.dir/all] Error 2 /home/ivy/workspace/catkin_ws_ov/devel/.private/ov_core/lib/libov_core_lib.so: undefined reference to `cv::aruco::detectMarkers(cv::_InputArray const&, cv::Ptrcv::aruco::Dictionary const&, cv::_OutputArray const&, cv::_OutputArray const&, cv::Ptrcv::aruco::DetectorParameters const&, cv::_OutputArray const&, cv::_InputArray const&, cv::_InputArray const&)'
> collect2: error: ld returned 1 exit status
> CMakeFiles/test_tracking.dir/build.make:219: recipe for target '/home/ivy/workspace/catkin_ws_ov/devel/.private/ov_core/lib/ov_core/test_tracking' failed
> make[2]: *** [/home/ivy/workspace/catkin_ws_ov/devel/.private/ov_core/lib/ov_core/test_tracking] Error 1
> CMakeFiles/Makefile2:899: recipe for target 'CMakeFiles/test_tracking.dir/all' failed
> make[1]: *** [CMakeFiles/test_tracking.dir/all] Error 2
> Makefile:140: recipe for target 'all' failed
> make: *** [all] Error 2`
> 
> I'm trying to build open vins, but it fails. How can I fix it?

goldbattle, 2019-10-29:
It looks like it is unable to find the aruco module.
- Did you build opencv with the contrib modules?
- What version of opencv are you using?

See this opencv install guide:
https://docs.openvins.com/gs-installing.html#gs-install-opencv

bckim1318, 2019-10-30:
I installed opencv follwing installation guide.

git clone --branch 3.4.6 https://github.com/opencv/opencv/
git clone --branch 3.4.6 https://github.com/opencv/opencv_contrib/
mkdir opencv/build/
cd opencv/build/
cmake -DOPENCV_EXTRA_MODULES_PATH=../../opencv_contrib/modules ..
make -j8
sudo make install

And I can import aruco module in python, so I think contrib module are installed well.

goldbattle, 2019-10-30:
From your error it looks like it can't find the aruco module.
I would check what version of opencv the ov_core library finds, you might have to specify the specific version if you have multiple installs on your system.

closed 2019-11-10 by bckim1318

Jin-Linhao, 2020-01-16:
I modify the CMakeLists to find_package(OpenCV 3.4 REQUIRED) and it works.

meitiever, 2020-02-17:
Did a little test on this issue, OpenCV 3.2 is not compatible with ov. 3.3.1 is ok.