rpng/open_vins #26: How do I use my own datasets?

yuhuabing, 2019-12-04:
When I use my own datasets,the terminal appears following error:
[ERROR] [1575476563.620017963]: cv_bridge exception: [8UC1] is not a color format. but [mono8] is. The conversion does not make sense.
I don't know how to solve it,could you help me,thank you!

goldbattle, 2019-12-04:
It looks like you are subscribing to a topic that is published as type `8UC1`.
I believe this is normally if the image is a depth map.
You will need to either manually convert it, or change your driver to publish in a good format.
Post and example bag file if you are able to.

NikolausDemmel, 2019-12-04:
Yes, it is behaviour of cv_bridge to refuse converting converting the `8UC1` format to intensities (even if the underlying data type is the same as `mono8`), see for example https://github.com/ros-perception/vision_opencv/issues/175.

So yeah, the source needs to be fixed, or possibly open_vins could be patched to do the conversion on the fly before calling cv_bridge, if you want that.

yuhuabing, 2019-12-05:
@goldbattle ,I just replace all "mono8" in source code with "TYPE_8UC1",it can work but always drift.

goldbattle, 2019-12-05:
It is an odometry method, so there will be drift over time.
SLAM features can help reduce this, please see the example launch files for
the euroc mav.
https://docs.openvins.com/gs-datasets.html

You can also post an example bag and your launch file and I can see if
there are any issues.

yuhuabing, 2019-12-05:
@goldbattle  could github upload bag file?After my modification,the system always fail to initialization.

goldbattle, 2019-12-05:
No, you would need to upload to Google drive or some other cloud service.

goldbattle, 2019-12-16:
Please reopen if you wish to troubleshoot any further, thanks!

closed 2019-12-16 by goldbattle