rpng/open_vins #27: Visualization about camera calibration

syywh, 2019-12-11:
Hi, thanks for opening this project. It runs well in my self-collected dataset !

But I found a small visualization problem when showing the camera calibration on RViz. Perhaps the codes in https://github.com/rpng/open_vins/blob/master/ov_msckf/src/core/RosVisualizer.cpp #Line261 & 268 should be modified since the quaternion representation used in your code is not the same as in tf::Quaternion?

goldbattle, 2019-12-13:
My understanding is that the TF transform contains: R_ItoG and p_IinG.

In our case, our state has R_GtoI and p_IinG, which means our rotation is the opposite of the one that is expected by ros (or the normal convention if using an SE(3) transformation). Since our quaternion is a JPL quaternion and Eigen uses Hamilton instead of performing an inverse, we can just send in our JPL quaternion and it should mean R_ItoG.

If you could expand a bit on the issue, that would be great.

syywh, 2019-12-16:
The visualization of the trajectory is correct. What I mean is the tf visualization of the camera extrinsics. I directly run the launch file in the "Simple Tutorial" session, and the visualization of the camera extrinsics is incorrect. There is no influence on the final IMU trajectory results, just the rotation between IMU and camera for tf visualization.

goldbattle, 2019-12-16:
Thanks! Took a while to realize the issue, thanks again for reporting.
Should be fixed in the latest commit to the repo.

commit 58e849e5 (2019-12-16): Fixed - Bug in calibration visualization (closes #27)
closed 2019-12-16 by goldbattle