rpng/open_vins #28: anchor frame time problem

HeadReaper-hc, 2019-12-14:
You said set the first measurement to be the anchor frame,but in the FUNCTION single_triangulation`feat->anchor_clone_timestamp = feat->timestamps.at(feat->anchor_cam_id).back(); ` you set the newest time, Am I not understanding correctly?

goldbattle, 2019-12-14:
Yes, we do this since if we initialize a feature in the oldest frame, right away we need to perform an anchor change since that old clone will be marginalized. We do loop through and find which camera has the most measurements, and use that frame to anchor the feature in.

HeadReaper-hc, 2019-12-15:
> Yes, we do this since if we initialize a feature in the oldest frame, right away we need to perform an anchor change since that old clone will be marginalized. We do loop through and find which camera has the most measurements, and use that frame to anchor the feature in.
thank you for your reply

closed 2019-12-15 by goldbattle