rpng/open_vins #32: Reading bag file differently

ravindraji4, 2020-01-27:
Thank you very much for prompt reply for issue  https://github.com/rpng/open_vins/issues/31.
I appreciate your help!! 

I have one more question: I would like to read the **V1_01_easy.bag** in separate terminal using rosplay for example. `rosbag play V1_01_easy.bag.bag --pause -s 10`. 

to achieve this, which file should be modified? an example would be really helpful.  
Looking forward to hear from you. Thank you very much.

goldbattle, 2020-01-27:
The difference between [pgeneva_eth.launch](https://github.com/rpng/open_vins/blob/master/ov_msckf/launch/pgeneva_eth.launch) and [pgeneva_eth_ros.launch](https://github.com/rpng/open_vins/blob/master/ov_msckf/launch/pgeneva_eth_ros.launch) is that the first one reads the bad in serially, thus you can change the start time etc. from inside the ros node. This *cannot* be used to subscribe to a live sensor since it needs to read a rosbag. This node is a nice way to debug the algorithm without worrying about it running real-time.

I believe you are looking to use the second one, so this plays back the bag on ros, and the node subscribes to the specified topics (as compared to reading the bad in serial). If you comment out line [98](https://github.com/rpng/open_vins/blob/master/ov_msckf/launch/pgeneva_eth_ros.launch#L97-L98) you will need to play the rosbag from another terminal, or subscribe to a real sensor as right now this launch file starts playing back the bag automatically.

ravindraji4, 2020-01-27:
@goldbattle Thank a lot, exactly that I was looking. It worked. 

To use Monocular vision. we will insert value = false at line 25 as following   `<param name="use_stereo"  type="bool"   value="false" />`?

goldbattle, 2020-01-27:
That is only if there is more than 2 cameras specified in `max_cameras`, so you don't need to change that. If you have two stereo images and want to process them in a binocular way (i.e. feature tracks are independent, no cross tracks) you can set the `use_stereo` to false.

ravindraji4, 2020-01-27:
@goldbattle  Thank a lot!!

closed 2020-01-27 by ravindraji4