rpng/open_vins #34: outlier rejection

moshanATucsd, 2020-01-31:
Hi, if I understand correctly, there's no gating test (eq. 6 in [this paper](http://www.roboticsproceedings.org/rss08/p31.pdf)) in your code. The only outlier rejection is the ransac used in tracking, is that correct? I am wondering whether there's a reason why there's no gating test used in this codebase?

goldbattle, 2020-02-01:
For filters it is pretty standard practice to always do a chi2 check before update.
We do this for all measurements see: [UpdaterMSCKF.cpp](https://github.com/rpng/open_vins/blob/master/ov_msckf/src/update/UpdaterMSCKF.cpp#L170-L200) and [UpdaterSLAM.cpp](https://github.com/rpng/open_vins/blob/master/ov_msckf/src/update/UpdaterSLAM.cpp#L308-L332)
Simplified the check is as follows:
```
// Chi2 distance check
Eigen::MatrixXd P = StateHelper::get_marginal_covariance(state, Hx_order);
Eigen::MatrixXd S = H_x* P*H_x.transpose() + R;
double chi2 = res.dot(S.llt().solve(res));
// Check against our threshold
if(chi2 > _options.chi2_multipler*chi2_check) {
    // chi2 check failed
} else {
    // chi2 check succedded
}
```

closed 2020-02-01 by moshanATucsd

SubMishMar, 2021-02-07:
@goldbattle this is a generic question about the chi2 test done in the update step. Why do you set the no of degrees of freedom of the chi2 test as 500?

goldbattle, 2021-02-08:
500 is just how many we cache for fast access.
We should index it by res.rows() which is the number of dof the measurement
has.

SubMishMar, 2021-02-08:
So, for a pixel measurement it should be 2? And for a position measurement (X, Y, Z) it should be 3? Just doing a sanity check.

nmerrill67, 2021-02-08:
@SubMishMar The residual contains multiple stacked 2-dimensional vectors for the uv reprojection error. So for example if we are processing 10 feature tracks, the dimension of the residual will be 20. The residual calculation can be seen in `ov_msckf/src/update/UpdaterHelper.cpp` in the `get_feature_jacobian_full` function.