rpng/open_vins #35: What type of hardware is being used?

jshortall, 2020-02-24:
I see from your documentation that you are using camera/IMU but don't see info about hardware. What are you using and calibrating for your tests?

https://docs.openvins.com/gs-calibration.html

Thanks for the info!

goldbattle, 2020-02-24:
We have run with a few different visual inertial sensors:
* Custom [pointgrey](https://www.flir.com/products/blackfly-s-usb3/) and [xsens](https://www.xsens.com/products/mti-100-series)
* [Intel Realsense](https://www.intelrealsense.com/tracking-camera-t265/) (t265 and original zr300)
* [Skybotix VI-Sensor](https://github.com/ethz-asl/libvisensor) (not sold any more)

We mainly just double sided tape the IMU to the pointgrey camera and follow the calibration procedure in that link you have posted. This page might also help you if you want to find a sensor:
https://github.com/ethz-asl/mav_tools_public/wiki/Visual-Inertial-Sensors

goldbattle, 2020-02-27:
Reopen if you have any more questions!

closed 2020-02-27 by goldbattle