rpng/open_vins #36: Issues in computing the length of the trajectory

ZuoJiaxing, 2020-02-28:
The function to compute the length of the trajectory is wrong:
https://github.com/rpng/open_vins/blob/4e0bd8105d8c9d2ba0cde9525f78ebe7d6e2094e/ov_eval/src/utils/Loader.cpp#L256

goldbattle, 2020-02-28:
The norm of the quaternion should be one, so this should be fine if as is.
It should probably be changed to explicitly norm just the position component for clarity.
Is this the issue that you are referring to?

goldbattle, 2020-02-28:
The norm of the *difference* of two quaternions is not still restrained to be one. Fixed in the latest commit. Reopen if this was not the issue. We would appreciate it if you provide more details of the error in the future, thanks!

commit 75ea024b (2020-02-28): Fixed #36 - Don't norm the quaterion for calculating trajectory length
commit 235bb881 (2020-02-28): Fixed #36 - Don't norm the quaterion for calculating trajectory length
closed 2020-02-28 by goldbattle