rpng/open_vins #39: About the FEJ implementation

liweizju, 2020-03-07:
Hi,
In Propagator.cpp, when do_fej, why do you use line280 instead of line281 (and similarly 283/284, 288/289, 294/295)? I think the line 281 is the correct one according to derivative of FEJ methods.

Thanks : )

goldbattle, 2020-03-07:
For the first one  (i.e. the change in angular velocity bias) we found that there wasn't that much difference between the two implementations in terms of performance. The bottom one should be more proper so feel free to try it out.
```cpp
F.block(th_id, bg_id, 3, 3).noalias() = -dR * Jr_so3(-w_hat * dt) * dt;
//F.block(th_id, bg_id, 3, 3).noalias() = -dR * Jr_so3(-log_so3(dR)) * dt;
```

For the last two, they should be equivalent if you use the property that R\*skew(v)\*R^t = skew(R\*v):
```cpp
F.block(v_id, th_id, 3, 3).noalias() = -skew_x(new_v-v_fej+_gravity*dt)*Rfej.transpose();
//F.block(v_id, th_id, 3, 3).noalias() = -Rfej.transpose() * skew_x(Rfej*(new_v-v_fej+_gravity*dt));
F.block(p_id, th_id, 3, 3).noalias() = -skew_x(new_p-p_fej-v_fej*dt+0.5*_gravity*dt*dt)*Rfej.transpose();
//F.block(p_id, th_id, 3, 3).noalias() = -0.5 * Rfej.transpose() * skew_x(2*Rfej*(new_p-p_fej-v_fej*dt+0.5*_gravity*dt*dt));
```

liweizju, 2020-03-08:
Got it, thanks!

closed 2020-03-08 by liweizju