rpng/open_vins #40: feature id in simulation

moshanATucsd, 2020-03-14:
Hi, 

In `Simulator::get_next_cam` in `src/sim/Simulator.cpp`, this part modifies the feature ids 

```
        // Append the map size so all cameras have unique features in them (but the same map)
        // Only do this if we are not enforcing stereo constraints between all our cameras
        for (size_t f=0; f<uvs.size() && !use_stereo; f++){
            uvs.at(f).first += i*featmap.size();
        }
```

in `feed_measurement_simulation` in `/src/track/TrackSIM.cpp` the ids are 

```
        // Update our feature database, with theses new observations
        // NOTE: we add the "currid" since we need to offset the simulator
        // NOTE: ids by the number of aruoc tags we have specified as tracking
        for(const auto &feat : feats.at(i)) {

            // Get our id value
            size_t id = feat.first+currid;
```

it seems that each feature measurement in one frame has a unique feature id, so I am wondering how does the tracker know that the measurements across different frames are from the same feature?

goldbattle, 2020-03-14:
In the first loop, the index `i` refers to the camera id.

In the case that we are not using stereo, then we will give each camera a unique map.
You can think of this as the first camera (camid=0) gets the generated map and its ids.
Then the second camera (camid=1) gets the same generated map, but all its ids will be the map size plus the original ids.

In the case that we are using stereo, then the same feature projected into all cameras will get the same id, while if not using stereo they will all have unique tracks. These ids are of course independent of time since we have a static map generated on startup. Hope this helps.

moshanATucsd, 2020-03-15:
Hi, thanks for the reply! I am still not quite clear how does the simulation generates a consistent feature track. For example, the first feature, with id = 1, can be observed at camera id 1, 2, suppose the uvs are u1, v1, u2, v2. Then the track is (1, [u1, v1]), (1, [u2, v2]), is this correct? The code above seems to generate a different feature id for each observation in each frame.

goldbattle, 2020-03-15:
It will generate a different feature ID for each **camera** for a given time instance if the stereo flag is not set. Your example is correct. In the case that we are not doing stereo constraints, then you will have (1, [u1, v1]), (1+mapsize, [u2, v2]).

closed 2020-03-15 by moshanATucsd