rpng/open_vins #42: OpenVINS standalone build

apache8080, 2020-03-18:
I am doing work with OpenVINS that requires it to be used without ROS. I have been able to hack around the build process to use OpenVINS in this way but I won't be able to pull updates from the upstream repo without a lot of manual work. I was wondering if you guys have considered building OpenVINS in a way that even the core libraries don't rely on ROS. Much of the core libraries are built in that way from what I have seen but they still `#include ros/ros.h` for the logging abilities. I think if OpenVINS were to have a standalone build to use as a library it would make it more widely used in a variety of different areas.

goldbattle, 2020-03-18:
We have a version which can build without ROS in the pipeline. Should be out in a few weeks. I will leave this issue open until I push that version public.

goldbattle, 2020-03-27:
Please checkout PR https://github.com/rpng/open_vins/pull/45, if you are able to test and provide feedback, I would be very grateful.

closed 2020-04-03 by goldbattle