rpng/open_vins #43: Discrete noise covariance matrices

AlessandroFornasier, 2020-03-24:
I am working with OpenVins and I have noticed something that is unclear for me, in particular in "ov_msckf/src/state/Propagator.cpp" at lines 327:328 reported below:

`Qc.block(6,6,3,3) = _noises.sigma_wb_2/dt*Eigen::Matrix<double,3,3>::Identity();
Qc.block(9,9,3,3) = _noises.sigma_ab_2/dt*Eigen::Matrix<double,3,3>::Identity();`

based on what reported in "Indirect Kalman Filter for 3D Attitude Estimation Technical Report
Number 2005-002, Rev. 57 March 2005" an on my knowledge I would like to say that the division "/" should be substituted by a multiplication "*" for the biases as follows:

`Qc.block(6,6,3,3) = _noises.sigma_wb_2*dt*Eigen::Matrix<double,3,3>::Identity();
Qc.block(9,9,3,3) = _noises.sigma_ab_2*dt*Eigen::Matrix<double,3,3>::Identity();`

Thanks.

goldbattle, 2020-03-24:
I would take a look at this issue first and see if that answers your question: https://github.com/rpng/open_vins/issues/14
Let me know if you need anything clarified or issues.

AlessandroFornasier, 2020-03-25:
Oh thanks, I got it now. Basically you divide by dt every entries of the Q matrix but on GQG^T the entries related to the biases will be multiplied by dt^2 restoring at the end the correct form _noises.sigma_xb_2*dt.

Thanks a lot for the support.

closed 2020-03-25 by AlessandroFornasier