rpng/open_vins #48: compare variable(type of double)

amosleeday, 2020-04-01:
A trivial question puzzled me.
Lots of  comparison of two double type variables in open_vins, like:
if (timefeat == timestamp) { //timefeat and timestamp are double
    has_timestamp = true;
    break;
}
However I didn't find the definenation of operator “==”. 
And in some std::map usage, double is the key. I didn't find the hashfunction of < or == either.
Maybe I lost important details, could you please tell me where the implementation is.

goldbattle, 2020-04-01:
These are language level comparison operators. In our case since we only ever copy the double timestamps, we don't have to necessarily worry about [floating point errors](https://randomascii.wordpress.com/2012/02/25/comparing-floating-point-numbers-2012-edition/) as compared to if we were changing the timestamps. Thus the `==` will require *exact* matching of the floating point numbers as compared to other ways which compare with some epsilon.

As for usage of maps, we use the [.find](http://www.cplusplus.com/reference/map/map/find/) on the keys which is efficient and provides O(logn) access time. The main reason we use maps is for the fast lookup and access if we already know the key, which is commonly the case, in O(1) time. The `==` in the case of the map should be on the interators which check if the iterator is the same (i.e. if the current iterator of the loop is equal to the iterator which marks the end of the map object).

amosleeday, 2020-04-02:
Thanks, Patrick. It helps a lot!

closed 2020-04-02 by goldbattle