rpng/open_vins #49: What is the coordinate system for OpenVINS?

apache8080, 2020-04-03:
Does OpenVINS use the default [ROS coordinate system](https://www.ros.org/reps/rep-0103.html)? Also if I am using OpenVINS with the EUROC dataset are the values going to be in the same frame as the vicon ball on the drone?

goldbattle, 2020-04-03:
We say that the global frame of reference is as in the rep:
```
x forward
y left
z up
```

In terms of the EuRoC datasets, you will need to use the optimized groundtruths that they provide (you can also find them [here](https://github.com/rpng/open_vins/tree/master/ov_data/euroc_mav) for convenience). We publish in the IMU sensor frame, and thus this wouldn't be in the vicon body frame or camera frame of reference.

apache8080, 2020-04-03:
So since you publish in the IMU sensor frame is it the same ref as the IMU in the EuRoC dataset if we are using the the EuRoC dataset? [This is the image I am using to get the reference frames for EuRoC.](https://projects.asl.ethz.ch/datasets/lib/exe/fetch.php?cache=&media=platform.jpg)
So according to this if I am using the EuRoC dataset then X is up, Y is left, Z is forward?

goldbattle, 2020-04-03:
Yes it should be.
All frames are right handed frames.

closed 2020-04-10 by goldbattle