rpng/open_vins #51: IMU biases and noise

juanmed, 2020-04-09:
Hi,

Thank you for open-sourcing such an amazing tool. 

I  hoped I could get some advice.  I needed to explicitely get the noise/bias corrected estimates of the IMU angular velocity and linear acceleration. These corrected measurements are used in the Propagation step as described [here](https://docs.openvins.com/propagation.html#conti_prop). My question is what are the correct values I should be looking to get and what is the proper way to extract them into ROS topics? I had two ideas and would appreciate your comments.

1) Create a new member publisher in `RosVisualizer` class and use it to publish the gyroscope and accelerometer biases in the method `publish_state` by getting the bias values using calls like `state->imu()->bias_g()`. This, however, would require that I somehow subtract this value from the correct imu measurement, which would need some sort of synchronization. I would prefer to avoid this.

2)  Make the local variables `w_hat` and `a_hat` in lines [299](https://github.com/rpng/open_vins/blob/18116ca299b6e7290b4ff7031eb471886140be8a/ov_msckf/src/state/Propagator.cpp#L299) and [300](https://github.com/rpng/open_vins/blob/18116ca299b6e7290b4ff7031eb471886140be8a/ov_msckf/src/state/Propagator.cpp#L299)  a member of `Propagator` class, and obtain them from `RosVisualizer` through its `VioManager* app` member. I might need to create a method `get_propagator` in `VioManager`  and `get_w_hat`, `get_a_hat` in `Propagator`. Finally use a call like `_app -> get_propagator() -> get_w_hat()` from `RosVisualizer`. If this was sensible, should I extract `w_hat, a_hat` or `w_hat2, a_hat2`?

This diagram shows, in red, the previous ideas:

[image]

Finally, I made some changes to publish biases. I got the following results using the Euroc easy rosbag. The field `linear_acceleration` is in deed `acceleration_bias` and `angular_velocity` is in deed `gyroscope_bias`. I was wondering if the estimated gyroscope biases are what is to be expected, since are very different from accelerometer biases and very small.

[image]
A zoomed-in view:
[image]

Thank you!

goldbattle, 2020-04-10:
The gyroscope bias should be pretty stable over the whole trajectory. The accelerometer bias on the other hand, I found that it seems to vary quite a lot (which you have also shown).

As for publishing, remember that over the propagation period the biases are constant. What you could do is save a "corrected" `vector<IMUDATA> prop_data`  and access that in your RosVisualizer (remember that the last entry is just used to calculated dt and thus shouldn't be published). From what you explained you will be publishing at the camera rate, as compared to publishing all imu measurements that are used for propagation.
https://github.com/rpng/open_vins/blob/18116ca299b6e7290b4ff7031eb471886140be8a/ov_msckf/src/state/Propagator.cpp#L133

juanmed, 2020-04-11:
Thanks for your feedback.

I will try to implement it as you described.

closed 2020-04-11 by juanmed