rpng/open_vins #54: About the error state propagation

liweizju, 2020-04-23:
Hi Patrick,

Thanks for sharing the codes!

I read and understand the derivation about the error-state propagation. 

But I have a question about the difference between your derivation and the one described in Joan Sola's "Quaternion kinematics for the error-state Kalman filter". In Joan's paper, the error-state kinematics is derived by combining the kinematics of true and norminal state, and then the error state propagation is obtained through certain integration method.  This should also be correct, but the result (using Euler integration as a sample, picture below) is different from yours.
I'm wondering why this happens? THANKS!

[image]

goldbattle, 2020-04-23:
I have not fully gone through the document you referenced, but an important understanding is that we use JPL quaternions. There is a nice table in [that paper](https://arxiv.org/pdf/1711.02508.pdf) which provides a good summary:

[image]

I think ignoring this difference, what matters is that we define our orientation error of our state. Looking at our [documentation](https://docs.openvins.com/propagation.html#error_prop) our error is defined in the local frame as a perturbation on the left of a rotation from global to local.
[image]

But the next question you might have is how does this relate to our underlying quaternion error state? To do this we can take a look at the [Trawny quaternion tech report](http://mars.cs.umn.edu/tr/reports/Trawny05b.pdf) where we have the following small angle relations (eq. 70, 71):
[image]
[image]

This shows that even if we find the Jacobian of the above SO(3) expansion (as on the documentation website) it will coincide with the quaternion Jacobian. I think the main difference with Sola's paper is that it uses the Hamilton quaternion definition as compared to our JPL which is based on the original MSKCF / Trawny filters.

liweizju, 2020-04-25:
Thanks for reply!

Yes, quaternion conventions are different in Joan's report (Hamilton) and your documentation (JPL), which will cause different derivation results. 
But it seems that the different of the term related to delta bias in error-state transition matrix is not the case. See the red box in the pictures.

[image]
from OpenVINS doc

[image]
from Joan's doc

I read related documentations again, and my current understanding is that Joan's error state propagation shown above is an approximate integration version using 1-st order truncated series. If the closed form integration is adopted for the error state integration (Joan's doc Chapter B), the transition matrix should be the same with yours. 
Actually, Trawny quaternion tech report you mentioned, already told the process (using closed form integration for calculating the error state transition matrix). I didn't notice that before.

goldbattle, 2020-04-25:
Glad we could help.
Feel free to reopen if there are any more issues (or open a new issue if it is a new specific question).

closed 2020-04-25 by goldbattle