rpng/open_vins #57: ZV-UP feature

ydsf16, 2020-05-01:
In this video, https://www.youtube.com/watch?v=MCzTF9ye2zw&t=30s,
**ZV-UP i**s used when the vehicle is static, can you share some references or open the feature in this repo.

goldbattle, 2020-05-01:
This is something I plan to release down the line, and it is on the todo list. No timeline for releasing it, but if you wish to create your own you can detect based on chi2 with the imu and image feature disparity check.

ydsf16, 2020-05-01:
Thanks for your reply. What I am curious about is what to do after the detection of vehicle static? Is it simple to not propagate and update, or add some static observation functions? If it is the latter, what kind of observation should be added?

goldbattle, 2020-05-01:
We directly changed the state time (i.e. identity state transition matrix through propagation) and updated the state. We also skipped adding feature measurements so that the feature tracking can continue un-interrupted after the zero velocity update starts.

ydsf16, 2020-05-01:
Just for confirming
1. At the propagation state, use the identity state transition matrix?
2. In the state augmentation stage, do not clone new camera states?
3. In the update stage, actually no update?

WoosikLee2510, 2020-05-01:
There can be multiple ways to implement ZV-UP. One of our versions skips
propagation/augmentation and updates the state using IMU
measurements/discarding camera measurements.

Woosik Lee

*UNIVERSITY of DELAWARE*
Ph.D. Student of Mechanical Engineering
328 Spencer Lab, Newark, DE, USA, 19716-3140
*e*mail: woosik@udel.edu

ᐧ

ydsf16, 2020-05-01:
Got it, thanks a lot.

rpng-guest, 2020-07-08:
PR #79 should address this.

closed 2020-07-08 by rpng-guest