rpng/open_vins #58: EUROC dataset and IMU excitation

apache8080, 2020-05-02:
I have set up a system where a EUROC dataset reader mimics the timing of the sensors in the dataset and passes the data to VioManager similar to a ros subscriber system. VioManager is setup with the params in the [pgeneva_ros_eth.launch](https://github.com/rpng/open_vins/blob/master/ov_msckf/launch/pgeneva_ros_eth.launch) file. When I am running the V1_01_easy (Vicon Room Easy) it takes about 5 seconds before there is enough IMU excitation to initialize the MSCKF and then the poses seem fairly accurate. This makes sense to me since the drone is sitting still for the first 4.7 seconds of the dataset. How accurate is the IMU_threshold of 1.5 for the EUROC dataset and in a real system would you have to manually tune this excitation to get accurate poses?

goldbattle, 2020-05-02:
This is specific to the IMU sensor and how the user picks it up. The MAV datasets normally have more vibration due to the motors so I have normally seen higher thresholds. This is normally a one-time tuning for the sensor / system. For example the TUM-VI datasets don't really have a "jerk" in the beginning so it is harder to initialize on those datasets. As per #56 and #57 I plan to have it be able to initialize right away if the user desires when I find time.

apache8080, 2020-05-02:
Oh ok, would this be similar to the way VINS-mono does online calibration? Also side note have you guys considered setting up some sort of discord, gitter, or IRC for faster communications? It has worked well for a lot of open source projects.

goldbattle, 2020-05-02:
I assuming you are referring to structure-from-motion initialization where the sensor can be initialized in motion. This is something we want to add down the line also. What I was originally referring too is allowing the system to initialize its orientation on startup without having to wait for a jerk in the imu.

closed 2020-05-02 by goldbattle