rpng/open_vins #59: OpenVINS Drifting to Infinity on Euroc Medium

apache8080, 2020-05-03:
Using the following launch file, [pgeneva_serial_eth.launch](https://github.com/rpng/open_vins/blob/master/ov_msckf/launch/pgeneva_serial_eth.launch), with OpenVINS and running the V1_02_medium dataset the poses drift to infinity. Has anybody else ran into this issue?

goldbattle, 2020-05-03:
Are you initializing with groundtruth?
If so you probably need to change the bag start time.

It should always work from a bag_start of zero if you are not initializing
with groundtruth.
To do that you just comment out the path_gt parameter.

apache8080, 2020-05-03:
I am not initializing with ground truth and I am starting at the beginning of the bag.

apache8080, 2020-05-03:
I am also running without ROS.

apache8080, 2020-05-03:
Also when I use the configuration from [this launch file](https://docs.openvins.com/gs-tutorial.html) it doesn't drift to infinity but it drifts by around 200 meters.

apache8080, 2020-05-03:
I realized I had an issue with timing in my non ROS system.

closed 2020-05-03 by apache8080