rpng/open_vins #60: Propagator::select_imu_readings(): There are not enough measurements to propagate with 0 sec missing

apache8080, 2020-05-03:
So I have created a system that uses the non-ROS OpenVINS and sends IMU/Camera data from the dataset files similar to the ros example subscribing to the ROS bag. I seem to be having some timing issues that are resulting the following to be printed to stderr:
`Propagator::select_imu_readings(): There are not enough measurements to propagate with 0 sec missing
Propagator::select_imu_readings(): IMU-CAMERA time offset is likely messed up, check time offset value!!!
/tmp/workspace/catkin_ws_ov/src/open_vins/ov_msckf/src/state/Propagator.cpp on line 207`

This is causing the VIO to drift a lot. I'm a little confused as to what causes this issue to occur so it is hard for me to debug my system. If you guys could give insight as to why an issue like this would happen that would help me a lot.

WoosikLee2510, 2020-05-03:
That is likely happen when you feed camera measurement that is more at future time than imu measurements you have in your system.
I would check: 1. If you are buffing camera measurements correctly, and 2. history of imu and camera measurements you fed in the system.

apache8080, 2020-05-03:
@WoosikLee2510 Thanks for the insight. I was able to solve this issue.

I wanted to also note that this error appeared in the older version of OpenVINS but in the latest OpenVINS 2 master this error doesn't print even though it is occuring. @WoosikLee2510 @goldbattle

WoosikLee2510, 2020-05-03:
Thanks for the feedback. We will take a look

closed 2020-05-04 by WoosikLee2510