rpng/open_vins #61: Coordinate axes confusion

Zee2, 2020-05-05:
Hi,

This is in relation to issue #49 .

We are using the EUROC MAV dataset as an input for OpenVINS. However, we are struggling to get the coordinate systems correct in the output. I'll walk through the steps we have taken.

First, we have correctly (to the best of our knowledge) applied the camera intrinsics and extrinsics.

Second, the EUROC dataset has several coordinate systems. The IMU has the axes:

[image]

Most notably in this diagram, we have the IMU axes:

```
x - up
y - right
z - forward
```

We also have the camera axes, which are:

```
x - right
y - down
z forward
```

As stated in issue #49, the coordinate system that OpenVINS publishes is as follows:

```
x - forward
y - left
z - up
```

Our target coordinate system is the standard OpenGL coordinate system, as follows:

```
x - right
y - up
z - forward
```

In #49, it was stated that the pose that OpenVINS publishes is "in the IMU frame". First of all, we're not entirely sure what this means. **We are passing the IMU data from EUROC directly into OpenVINS without any transformation**, so, the data that is being passed into OpenVINS is according to the axes stated above as the IMU axes (x - up, y - right, z - forward). 

When examining the pose data that is published by OpenVINS, I attempted to transform the data from OpenVINS into the OpenGL coordinate system. I did this as follows:

```
gl_pose.x = -ov_pose.y;
gl_pose.y = ov_pose.z;
gl_pose.z = ov_pose.x;
```

This should map the coordinate system that OpenVINS uses to publish poses (as stated in #49) to the OpenGL coordinate system. I did a similar transformation for the orientation quaternion.

```
gl_orientation.w = ov_orientation.w;
gl_orientation.x = -ov_orientation.y;
gl_orientation.y = ov_orientation.z;
gl_orientation.z = ov_orientation.x;
```
Several issues have arisen from this mapping.

First of all, most clearly, the position obtained through the above pose remapping seems to have the x coordinate backwards. In the dataset, the drone clearly moves forward as the first motion; however, using this mapping, the position moves backwards. (All other axes seem correct, both "up" and "right" behave correctly.) When I make the "forward" axis negative, the published pose appears completely accurate.

Secondly, the published orientation behaves oddly; when the first orientation is first published by OpenVINS, it is entirely "weird": it is upside down, tilted at an odd angle, and just generally incoherent. _**However,**_ when I take this initial orientation as a sort of "offset quaternion", and apply this initial offset quaternion to all future orientations that OpenVINS emits, the orientation appears completely correct.

Thus, in summary:

We are feeding the IMU data from the EUROC dataset (as described above) into OpenVINS, unmodified. (We are also feeding in the camera data, unmodified.) We have applied the given camera extrinsics and intrinsics.

After receiving the position and orientation from OpenVINS, we apply the above re-mapping to transform the "OpenVINS coordinate space" into the standard OpenGL coordinate space.

Result: Position is correct, but one axis is reversed. Orientation is correct, but only after applying the first orientation as an offset, "origin" quaternion.

We would _greatly_ appreciate your help with this. This has been driving us up the wall for several weeks now.

Thanks,

Finn Sinclair + the ILLIXR team at University of Illinois at Urbana-Champaign.

Zee2, 2020-05-05:
Apologies. There is an internal inconsistency in the coordinate frames we are using, elsewhere in our application.

I'll reopen this issue in the future when we've cleared up the inconsistencies.

closed 2020-05-05 by Zee2

goldbattle, 2020-05-05:
I can comment that the initial position and yaw angle are pretty arbitrary. The initialization procedure can only determine the roll and pitch angles based on gravity. The position and yaw are then assigned (typically to zero), but can be assigned to anything without affecting the relative position estimate of the VIO system.

When I referred to the "imu sensor frame" that is the physical frame that is attached to the IMU. One could transform this into any other sensor frame that is rigidly attached to the sensor unit, but since we take the IMU as the base of the system that is what we publish. If you wish to get the camera, you can apply the estimated rigid transformation to find the pose in either the left or right camera as needed.

re: axis convention, this would just need to be done if you change from a right hand coordinate system to another, or for example the convention that z being the vertical direction is different (for example your opengl has that the vertical is not the z-axis by convention). Just note that this is not something specific to the sensor imu frame, but instead is the *global frame* of reference. You wish to transform from the global ROS standard to an opengl axis standard.

cwcentral, 2024-03-09:
Show me how to access the w variable of a Eigen::Quaternion