rpng/open_vins #62: RGBD sensor input

NicoNorena, 2020-05-06:
Are there any plans to support RGBD as an input? I have seen a modification of VINS-mono for RGBD, I was wondering if you are familiarized with this implementation and whether it could be similarly be applied to open_vins.

What are your thoughts?

goldbattle, 2020-05-07:
It is something we have considered and is an interesting extension. Some key issues to overcome are the depth sensor calibrated and noise model, along with how best to incorporate the information in a meaningful measurement model. We have done some [work](http://udel.edu/~pgeneva/downloads/papers/c05.pdf) which leveraged depth information in a plane+vio system. More specifically to this codebase, to add support we would want to include thorough documentation and online calibration support which is non-trivial. I think the main reason we wouldn't want to add it is lack of a good evaluation dataset which could facilitate research on. Do you know of any public datasets which provide visual/inertial/depth information?

NikolausDemmel, 2020-05-07:
Have you seen [ETH3D](https://www.eth3d.net/slam_overview)? Haven't used it myself, but sounds like what you want.

goldbattle, 2020-05-07:
Thanks for the link, it looks to be one of the only datasets out there with RGBD+IMU+GT. Looking at the raw data they provide the IR images from their Xtion Live Pro, but the processed datasets zips seem to have the depthmaps precomputed.

NicoNorena, 2020-05-07:
it looks like this another dataset  as rosbag, and the depth is precomputed on the RealSense D435i.
the left and right images from the RealSense are IR images (global shutter), and one RGB camera with a (rolling shutter).

`/camera/aligned_depth_to_color/image_raw       1866 msgs    : sensor_msgs/Image        
 /camera/aligned_depth_to_colorizer/image_raw    1866 msgs    : sensor_msgs/Image        
  /camera/color/image_raw                         1865 msgs    : sensor_msgs/Image        
   /camera/depth/image_rect_raw                  1871 msgs    : sensor_msgs/Image        
  /camera/imu                              30926 msgs    : sensor_msgs/Imu          
  /vrpn_client_node/handheld/pose               11599 msgs    : geometry_msgs/PoseStamped`

https://robotics.shanghaitech.edu.cn/datasets/VINS-RGBD

@goldbattle as you described, a dataset with the raw  depth data sensor is what is needed. I have not seen any public datasets with the raw data.  The BSP for the RealSense does not publish the raw data, it only publishes the processed data. Also, the IMU is sync with the frame rate such that one IMU measurement per frame. the raw IMU measurement can be published, but the gyroscope and accelerometer publish at two different frequencies in two different topics. Maybe the source can modified to capture the raw data, this might not be possible as some of the interaction with Intel devices is a black box.

I am wondering if as a first step leveraging  the RealSense D35i with the precomputed depth, can the plane+vio incorporate the meaningful information in the measurement model? 

I am guessing that a offline precalibration of  the IMU and depth  sensor is needed, on top of the online calibration. I am not familiarized with this calibration . 

if using the RealSense D435i, the RGB camera uses a rolling shutter, I have to double check this. If it is, adding this support is not something feasible.  Are there any of the shelf platform that uses an open shutter?

NicoNorena, 2020-05-14:
@goldbattle Is the MSCKF+point-on-plane code base already incorporate as part of OpenVINS? do you have it publicly available?

WoosikLee2510, 2020-05-14:
@NicoNorena we have some applications, but the application package is not in public yet. Hopefully open source soon?

NicoNorena, 2020-05-27:
cool, can you let me know when you do? I like to tested it. The paper was very interesting, it sounded like you guy used bare version of openvins

yangyulin, 2020-05-27:
If you refer to the point-on-plane paper[1], that system is based on an old
version of open-vins.
The current plan is to refine the system with line and plane in this fall.
If I can finish that on time, we need to go through group discussion to
decide when and how to release it.

[1]Yang 2019 ICRA, Tightly-coupled aided inertial navigation with point and
plane features
<https://scholar.google.com/scholar?oi=bibs&cluster=3129610941112037607&btnI=1&hl=en>

suho0515, 2020-06-24:
i made rgbd camera version of open_vins.
take a look at it, if you interested.

https://github.com/suho0515/open_vins_rgbd

i used VINS_RGBD datasets.

NicoNorena, 2020-07-05:
@suho0515 thanks, I will take a look and test it out.

Is this fork implementation based on the paper "Tightly-Coupled Aided Inertial Navigation
with Point and Plane Features" ?

suho0515, 2020-07-06:
> @suho0515 thanks, I will take a look and test it out.
> 
> Is this fork implementation based on the paper "Tightly-Coupled Aided Inertial Navigation
> with Point and Plane Features" ?

no, i didn't reffered any paper.
it was simple modifying from ov_msckf source.
i tried to keep the structure of "ov_msckf". :)

closed 2020-11-30 by goldbattle

NicoNorena, 2020-12-08:
@goldbattle I was wondering if you guys were going to add RGBD support to this codebase? you mentioned about supporting the work on plane+vio system to this codebase.

goldbattle, 2020-12-08:
At this point in time we don't plan to add this support.

MMOCKING, 2022-01-04:
> no, i didn't reffered any paper. it was simple modifying from ov_msckf source. i tried to keep the structure of "ov_msckf". :)

> i made rgbd camera version of open_vins. take a look at it, if you interested.
> 
> https://github.com/suho0515/open_vins_rgbd
> 
> i used VINS_RGBD datasets.

Do you have any dataset with groundtruth? VINS_RGBD datasets does not appear to have groundtruth data...

suho0515, 2022-01-04:
> Do you have any dataset with groundtruth? VINS_RGBD datasets does not appear to have groundtruth data...

I didn't make other dataset version, but you can just adjust topic name from the other dataset.
It's been quite long time to touch the repository, so I'm not sure how to adjust it for your case.
But if you have any other question I'll try to look at it for you :)