rpng/open_vins #68: How about change IMU calibration model to improve the precision of VIO

Wangxuefeng92, 2020-05-21:


WoosikLee2510, 2020-05-21:
Can you suggest what model you have in your mind? Is it IMU intrinsic or extrinsic? We recently published IMU intrinsic calibration method (Analytic Combined IMU Integration for Visual-Inertial Navigation, ICRA2020) so this might be what we will add in our system in the future.

yangyulin, 2020-05-21:
Thanks for your interest in OpenVINS.
Our colleagues have given you some feedback regarding your questions above.
I will just give some additional comments.

1) Regarding the IMU calibration model, do you refer to the IMU noise model
(like how to model biases) or IMU intrinsics model?

2) If for the IMU noise model, OKVIS used a slightly different bias model
(at least from their paper). I have no experience on that bias model yet,
but I guess it won't change the accuracy much.
http://www.roboticsproceedings.org/rss09/p37.pdf

3) If for the IMU intrinsic model, we have a recent paper on adding IMU
intrinsic calibration to OpenVINS with 4 model variants.
It can improve the system performance *only when you can fully excite the
6D motion of the IMU. *We find out that the IMU intrinsic calibration is
very sensitive to motions. The paper will come out soon in July @ RSS2020.

Wangxuefeng92, 2020-05-22:
thank you for your replies!

closed 2020-05-22 by Wangxuefeng92