rpng/open_vins #69: about accounting for the uncertainty of time offset in augmenting covariance

JzHuai0108, 2020-05-28:
Is [this line](https://github.com/rpng/open_vins/blob/master/ov_msckf/src/state/StateHelper.cpp#L577-L578) necessary?

kevineck, 2020-05-28:
Please see closed issue #13 for the derivation.

goldbattle, 2020-05-28:
I believe you are correct, as the code operates sequentially on the same matrix. The covariance will not be inconsistent, but just not as confident as possible. The second line here will already add the H\*P\_tofftoff\*H^T since it is working on what the first line calculated.
```
state->_Cov.block(0, pose->id(), state->_Cov.rows(), 6) +=
     state->_Cov.block(0, state->_calib_dt_CAMtoIMU->id(), state->_Cov.rows(), 1) * dnc_dt.transpose();
state->_Cov.block(pose->id(), 0, 6, state->_Cov.rows()) +=
     dnc_dt * state->_Cov.block(state->_calib_dt_CAMtoIMU->id(), 0, 1, state->_Cov.rows());
state->_Cov.block(pose->id(), pose->id(), 6, 6) +=
     dnc_dt * state->_Cov(state->_calib_dt_CAMtoIMU->id(), state->_calib_dt_CAMtoIMU->id()) * dnc_dt.transpose();
```
Thus there shouldn't be a need to add the H\*P\_tofftoff\*H^T at the end. This is probably from legacy code where we worked on a **copy** of the covariance in which this code above will be correct. Thanks for the close inspection.

JzHuai0108, 2020-05-31:
Thank you for the clarification. I would not mind if you close this issue for now.

commit 2295e7f8 (2020-07-06): Fix additional inflation of toff during augmentation (fixes #69)

The original code was for inplace covariance operations.
This caused an additional H*P_tofftoff*H^T to be added.
While this shouldn't hurt it just means our covariance is conservative after augmentation.
Removing doesn't seem to change simulation performance.
closed 2020-07-06 by rpng-guest