rpng/open_vins #70: Is it a necessary check in Propagator::SelectImuReadings() ?

meitiever, 2020-06-09:
Code as below:
`if (imu_data.at(imu_data.size() - 1).timestamp <= time1) {  
    LOG(ERROR) << "Propagator::SelectImuReadings(): There are not enough measurements to propagate with " << (time1 - imu_data.at(imu_data.size() - 1).timestamp) << " sec missing"; 
    LOG(ERROR) << "Propagator::SelectImuReadings(): IMU-CAMERA time offset is likely messed up, check time offset value!!!";  
    LOG(ERROR) << __FILE__ << " on line " << __LINE__;  
    return prop_data;  
}`
disable time offset calibration, Is this a necessary check? 
if the system is strictly synchronized, at some time point, the imu data ts is the same with image ts exactly and then the error will fire up.
such as imu measurements ts: 1,2,3,4,5,6,7,8,9,10 and vio manager received these 6 imu data.
and then image measurement with ts 5, 10 sent to vio manager. 
the first image is fiine but the second image with ts 10 will fire up the error.
the  < check is necessary, but the = check is not.

WoosikLee2510, 2020-06-09:
Briefly answering, we cache one image measurement so that we have enough imu measurements to wrap the image meas. By this, we can handle the time offset and the asynchronocity of the camera measurement.

WoosikLee2510, 2020-06-09:
So if you have "perfect" synchronization between the sensors and guarentee no msg drops, then yes you don't need this check. However, most of the real world cases there are some intermittency and hiccups in sensor meas, so we have img caching and imu meas check for safety.

meitiever, 2020-06-09:
Thanks for your response so fast.
different from the real device, I tested my data over ros websocket currently.
before open vins, the synchronization flow gave me a data structure contain 2 images with the same ts and some imu data up to this ts.
when I feed these data to vio manager, I have got a lot of "not enough measurements" errors.
seems that i have to feed more imu data over the ts to meet this if statement.
but it is impossible for me(make me to change the sync flow). the best i can get is equal. the last imu ts is equal to the image ts in this update loop.
so i remove the = in the if statement.
could you give me some idea of how to fit this data structure into open vins?
thanks.

goldbattle, 2020-06-09:
I believe this isn't in the master branch any more?
Which branch are you using?
The one that I see now is warning that you need at least two IMU measurements to propagate with (since you need two to calculate a dt time difference between camera times).
https://github.com/rpng/open_vins/blob/master/ov_msckf/src/state/Propagator.cpp#L276-L278

You get two image measurements with the same timestep?
As in they are stereo cameras?
Then you should send in using the feed_stereo interface.

If this is monocular, then you will need to address the issue in your driver, or drop the second image that has the same timestep. Having two images at the same timestep doesn't make sense since a single camera can't physically collect two images at the same time.

meitiever, 2020-06-10:
I cloned the master branch a few months ago and merged into my solution.
yes, I capture images from the stereo camera and the two cameras are synchronized, so the time stamp are the same.
I updated the master branch today and will do more tests.
feed back this week.
thanks.

meitiever, 2020-06-17:
the lastest code remove this warning.
but i found an issue of adopting the data pattern to open vins.
SelectImuReadings() method needs some changes.

WoosikLee2510, 2020-06-17:
@meitiever could you be more specific about the pattern you are mentioning and what changes suggesting?

goldbattle, 2020-06-26:
Please reopen / post details if you get back to this. Thanks,

closed 2020-06-26 by goldbattle