rpng/open_vins #71: About Feature triangularization

canyou2014, 2020-06-15:
this line: (https://github.com/rpng/open_vins/blob/master/ov_core/src/feat/FeatureInitializer.cpp#L263)

why is  `Hess_l(r,r) *= (1.0+lam)`, not `Hess_l(r,r) += lam;` ?
while traditional LM is this form: ` (A + lam * I) = b`, it can keep the matrix positive.

goldbattle, 2020-06-15:
I believe we use the modified form.
See eq 13 here for the form that I believe we use:
http://people.duke.edu/~hpgavin/ce281/lm.pdf

Section 2.2 of this paper also has a good discussion and references / more
details.
https://arxiv.org/pdf/1201.5885.pdf

Wikipedia also has a note of it in the first section:
https://en.wikipedia.org/wiki/Levenberg%E2%80%93Marquardt_algorithm

canyou2014, 2020-06-20:
OK, Thank you.

closed 2020-06-20 by canyou2014