rpng/open_vins #72: is there special reason for choosing param for "cv:findFundamentalMat()"?

suho0515, 2020-06-16:
hello sir,
I'm Suho, university student from south korea.

first of all, thanks for your great work.

when i look around your work, i could find that you guys using max focal length for choosing parameter for "cv::findFundamentalMat(pts0_n, pts1_n, cv::FM_RANSAC, 1/max_focallength, 0.999, mask_rsc);".

is there special reason for using "1/max_focallength"?

it's just pure curiosity.
so let me know the logic whenever you have a enough time.

again, thank you for your amazing works. :)

goldbattle, 2020-06-17:
Here is the documentation for the function:
https://docs.opencv.org/3.4.10/d9/d0c/group__calib3d.html#ga59b0d57f46f8677fb5904294a23d404a

> ransacReprojThreshold
> Parameter used only for RANSAC. It is the maximum distance from a point to an epipolar line in pixels, beyond which the point is considered an outlier and is not used for computing the final fundamental matrix. It can be set to something like 1-3, depending on the accuracy of the point localization, image resolution, and the image noise.

This is the threshold used to determine if a given pixel is an inlier or outlier after reprojection from frame 1 to frame 2. If we were using the raw pixels then we would say that we would expect 1-3 pixel reprojection error for features (which is what the opencv documentation is referring too).

In our case, we have normalized the uv coordinates, thus this doesn't not correspond to a raw pixel in the image. So we can directly transform the 1 pixel raw image error into the normalized coordinates by dividing by the focal length. We transform into the normalize coordinates, as for when you have distortion the fundamental matrix won't hold for all features.

suho0515, 2020-06-18:
> Here is the documentation for the function:
> https://docs.opencv.org/3.4.10/d9/d0c/group__calib3d.html#ga59b0d57f46f8677fb5904294a23d404a
>
>
> This is the threshold used to determine if a given pixel is [...]

thanks! it was very understandable ! :)

closed 2020-06-18 by suho0515