rpng/open_vins #73: are these lines necessary in "update_feature()"?

suho0515, 2020-06-18:
hello sir,
first of all, thanks for your great works.
I was looking around your codes.
and got a little curiosity.

in the "update_feature()" function.

`        /**
         * @brief Update a feature object
         * @param id ID of the feature we will update
         * @param timestamp time that this measurement occured at
         * @param cam_id which camera this measurement was from
         * @param u raw u coordinate
         * @param v raw v coordinate
         * @param u_n undistorted/normalized u coordinate
         * @param v_n undistorted/normalized v coordinate
         *
         * This will update a given feature based on the passed ID it has.
         * It will create a new feature, if it is an ID that we have not seen before.
         */
        void update_feature(size_t id, double timestamp, size_t cam_id,
                            float u, float v, float u_n, float v_n) {

            // Find this feature using the ID lookup
            std::unique_lock<std::mutex> lck(mtx);
            if (features_idlookup.find(id) != features_idlookup.end()) {
                // Get our feature
                Feature *feat = features_idlookup[id];
                // Append this new information to it!
                feat->uvs[cam_id].emplace_back(Eigen::Vector2f(u, v));
                feat->uvs_norm[cam_id].emplace_back(Eigen::Vector2f(u_n, v_n));
                feat->timestamps[cam_id].emplace_back(timestamp);
                return;
            }

            // Debug info
            //ROS_INFO("featdb - adding new feature %d",(int)id);

            // Else we have not found the feature, so lets make it be a new one!
            Feature *feat = new Feature();
            feat->featid = id;
            feat->uvs[cam_id].emplace_back(Eigen::Vector2f(u, v));
            feat->uvs_norm[cam_id].emplace_back(Eigen::Vector2f(u_n, v_n));
            feat->timestamps[cam_id].emplace_back(timestamp);

            // Append this new feature into our database
            features_idlookup.insert({id, feat});
        }
`

there is lines like below.

`            if (features_idlookup.find(id) != features_idlookup.end()) {
                // Get our feature
                Feature *feat = features_idlookup[id];
                // Append this new information to it!
                feat->uvs[cam_id].emplace_back(Eigen::Vector2f(u, v));
                feat->uvs_norm[cam_id].emplace_back(Eigen::Vector2f(u_n, v_n));
                feat->timestamps[cam_id].emplace_back(timestamp);
                return;
            }
`

if there is same id, 
shouldn't we just ignore it? 
is there need for copying the data?

I mean...
is it okay to edit like below? 

`            if (features_idlookup.find(id) != features_idlookup.end()) {
                return;
            }
`

The program works greatly, but i just curious about the codes.

thanks for reading and hope your any advice of it. :)

goldbattle, 2020-06-18:
This function will either append a new observation at the passed timestep if there is already a feature created (which will happen as soon as you track a feature from frame one to frame two), or if it does not exist yet it will create a new feature storage object and append to that.

If you change it to what you propose, then each feature will only have its first ever track (uv and timestamp) in it.

The end goal is to collect a [Feature](https://docs.openvins.com/classov__core_1_1Feature.html) object which has all measurement information inside of it, and the corresponding image timesteps that each measurement was collected at. If you are interested, see how we interface with it in the [trackers](https://github.com/rpng/open_vins/blob/master/ov_core/src/track/TrackKLT.cpp#L107-L113), and then how we call it in the [viomanager](https://github.com/rpng/open_vins/blob/master/ov_msckf/src/core/VioManager.cpp#L357-L367).

suho0515, 2020-06-19:
> This function will either append a new observation at the passed timestep if there is already a feature created (which will happen as soon as you track a feature from frame one to frame two), or if [...]

thank you for your kindness,
i guess i miss something though...

`        if (features_idlookup.find(id) != features_idlookup.end()) {
            // Get our feature
            Feature *feat = features_idlookup[id];
            // Append this new information to it!
            feat->uvs[cam_id].emplace_back(Eigen::Vector2f(u, v));
            feat->uvs_norm[cam_id].emplace_back(Eigen::Vector2f(u_n, v_n));
            feat->timestamps[cam_id].emplace_back(timestamp);
            return;
        }`

in the codes above,
the pointer "Feature *feat" is created,
and values(u,v,u_n,v_n,timestamp) are put in "feat",
then return.

is that mean the "Feature *feat" pointer disappear?
then does we need to put all the variables to "feat" pointer?

i'm pretty beginner for programming, sorry for asking the basic question.

and thanks again for your kindness.

goldbattle, 2020-06-19:
A new pointer will not be created, it is grabbed from the database map.
Pointers are only created with the keyword "new" like below this if statment.

suho0515, 2020-06-19:
> A new pointer will not be created, it is grabbed from the database map.
> Pointers are only created with the keyword "new" like below this if statment.

now i can see what's happening! 
thank you very much ! :)

i'll close this issue then.

closed 2020-06-19 by suho0515