rpng/open_vins #75: pointcloud rotation problem.

suho0515, 2020-06-24:
hello sir, 
recently, i made rgbd version of open_vins.
and i made a custom dataset and did run the program.
and tried to make octomap using point cloud data.

first of all, localization seems perfect. (not perfectly sure, cause i don't have ground truth)

[image]
 
but octomap seems terrible.

[image]

i guess the problem is pointcloud's degree.
cause pointcloud allways turn around almost 15~30 degree.

[image]

i publish pointcloud in RosVisualizer.cpp and made "frame_id" to "cam0".

` 

// publish pointcloud from depth image
void RosVisualizer::publish_pointcloud() {

    // Check if we have subscribers
    if(pub_tracks.getNumSubscribers()==0 || poses_imu.size() < 1)
        return;

    // Get our pointcloud
    Map *map = _app->get_map();

    std::vector<Eigen::Vector3d> pointcloud = map->get_pointcloud();

    // Declare message and sizes
    sensor_msgs::PointCloud2 cloud;
    cloud.header.frame_id = "cam0";
    cloud.header.stamp = ros::Time::now();
    cloud.width  = 3*pointcloud.size();
    cloud.height = 1;
    cloud.is_bigendian = false;
    cloud.is_dense = false; // there may be invalid points

    // Setup pointcloud fields
    sensor_msgs::PointCloud2Modifier modifier(cloud);
    modifier.setPointCloud2FieldsByString(1,"xyz");
    modifier.resize(3*pointcloud.size());

    // Iterators
    sensor_msgs::PointCloud2Iterator<float> out_x(cloud, "x");
    sensor_msgs::PointCloud2Iterator<float> out_y(cloud, "y");
    sensor_msgs::PointCloud2Iterator<float> out_z(cloud, "z");

    for (const auto &pt : pointcloud)
    {

        *out_x = pt(0);
        ++out_x;
        *out_y = pt(1);
        ++out_y;
        *out_z = pt(2);
        ++out_z;
    }

    // Publish
    pub_pointcloud.publish(cloud);

}

`

i did camera-imu calibration using Kalibr.
Reprojection error was 0.4835, not perfect but looks reasonable. 
results are followed.

Calibration results
===================
Normalized Residuals
----------------------------
Reprojection error (cam0):     mean 0.483512585192, median 0.422313346817, std: 0.303837372257
Gyroscope error (imu0):        mean 0.639279430331, median 0.594986553131, std: 0.316159543726
Accelerometer error (imu0):    mean 0.923006921728, median 0.803086304199, std: 0.660446981304

Residuals
----------------------------
Reprojection error (cam0) [px]:     mean 0.483512585192, median 0.422313346817, std: 0.303837372257
Gyroscope error (imu0) [rad/s]:     mean 0.00167491210747, median 0.0015588647692, std: 0.000828338004563
Accelerometer error (imu0) [m/s^2]: mean 0.02370281775, median 0.0206232562918, std: 0.0169602784799

Transformation (cam0):
-----------------------
T_ci:  (imu0 to cam0): 
[[ 0.99974786 -0.0033015   0.02221071  0.0228376 ]
 [ 0.00336317  0.99999059 -0.00273984  0.00349851]
 [-0.02220145  0.00281384  0.99974956 -0.01651559]
 [ 0.          0.          0.          1.        ]]

T_ic:  (cam0 to imu0): 
[[ 0.99974786  0.00336317 -0.02220145 -0.02321028]
 [-0.0033015   0.99999059  0.00281384 -0.0033766 ]
 [ 0.02221071 -0.00273984  0.99974956  0.0160138 ]
 [ 0.          0.          0.          1.        ]]

timeshift cam0 to imu0: [s] (t_imu = t_cam + shift)
-0.025676385167939447

Gravity vector in target coords: [m/s^2]
[ 0.171924   -9.80450194 -0.10298927]

Calibration configuration
=========================

cam0
-----
  Camera model: pinhole
  Focal length: [598.9901005976366, 597.81854862048]
  Principal point: [307.69324783179036, 251.27228059956354]
  Distortion model: radtan
  Distortion coefficients: [0.09505626302907584, -0.09767276595601246, 0.0029487056339383474, -0.004662649985855509]
  Type: aprilgrid
  Tags: 
    Rows: 6
    Cols: 6
    Size: 0.025 [m]
    Spacing 0.0075 [m]

IMU configuration
=================

IMU0:
----------------------------
  Model: calibrated
  Update rate: 400.0
  Accelerometer:
    Noise density: 0.001284 
    Noise density (discrete): 0.02568 
    Random walk: 0.000209
  Gyroscope:
    Noise density: 0.000131
    Noise density (discrete): 0.00262 
    Random walk: 3e-06
  T_i_b
    [[1. 0. 0. 0.]
     [0. 1. 0. 0.]
     [0. 0. 1. 0.]
     [0. 0. 0. 1.]]
  time offset with respect to IMU0: 0.0 [s]

maybe the rotation pose would be wrong? but at least localization seems perfect.

have i missed something?
any clue would be appreciated.

goldbattle, 2020-06-25:
Are you sure they are in the same frame of reference?

Not sure what sensor you use, but normally it will be in whatever depth sensor frame the data is collected in. So you would need to publish the pointcloud in the depth0 frame and also publish the cam0 to depth0 transform. I know that some of the intel products will produce an "aligned with rgb depth frame" which will be in the same frame, but this is not the normal case.

suho0515, 2020-06-25:
> Are you sure they are in the same frame of reference?
>
> Not sure what sensor you use, but normally it will be in whatever depth sensor frame the data is collected in. So you would need to publish the [...]

Thank you goldbattle.
I'm using Realsense D435i.
you are right, i might have to transform cam0 to depth0.

[image]

i guess "/camera/extrinsics/depth_to_color" topic has related with it.

eh420@eh420-Strix-GL704GW-GL704GW:~/catkin_ws_realsense$ rostopic echo /camera/extrinsics/depth_to_color 
header: 
  seq: 0
  stamp: 
    secs: 0
    nsecs:         0
  frame_id: "depth_to_color_extrinsics"
rotation: [0.9999950528144836, -0.0019471179693937302, -0.0024772195611149073, 0.0019419413292780519, 0.9999959468841553, -0.0020903833210468292, 0.0024812796618789434, 0.002085562329739332, 0.9999947547912598]
translation: [0.014739668928086758, 7.83317445893772e-05, 0.0005913236527703702]

there is also "aligned with rgb depth frame" option.

i'll try it and leave the result!

suho0515, 2020-06-26:
> Are you sure they are in the same frame of reference?
>
> Not sure what sensor you use, but normally it will be in whatever depth sensor frame the data is collected in. So you would need to publish the [...]

I'd add some result after solving the point cloud degree problem.

I edited "cam0" frame of point cloud to "depth0".
and add "static_transform_publisher" to connect "cam0" to "depth0".
i don't know why but the yaw(z axis) wasn't the axis about rotation, so i changed pitch(y axis) with value i got from experiments.

`
    <node pkg="tf"
        type="static_transform_publisher"
        name="cam0_to_depth0"
        args="0.0 0.0 0.0 0.0 0.39 0.0 cam0 depth0 100" />    
`

the result is below.

[image]

the result map looks pretty mass, i thought there are two reasons.

1. depth image noise
i think i should use some filtering to solve this issue.

2. inaccurate rotation pose when it rotate at corner. 
when sensor rotate at corner, the rotation poses are unstable. 
i guess that's why the map looks like windmill at corner .

i might sould work on this too, though there is no clue yet.

thank you very much about your help!
i'll close this issue then.

closed 2020-06-26 by suho0515