rpng/open_vins #76: No visualization

ShrutheeshIR, 2020-06-25:
Hello. Thanks for the amazing work!

I tried running the aruco marker example code using ```roslaunch ov_msckf pgeneva_serial_aruco.launch```. While it does work, and values are being computed, there is no visualization, even though the code calls visualization by default nothing happens. On the terminal it prints the different values at every iteration. However, at the end this is the output

`cam0 intrinsics:
461.370,462.763,405.499,246.132
-0.27247,0.06502,0.00068,-0.00003

cam1 intrinsics:
465.048,466.608,385.196,269.340
-0.27640,0.06930,0.00064,0.00012

T_C0toI:
-1.000,-0.005,0.001,0.041,
0.005,-1.000,0.005,-0.010,
0.001,0.005,1.000,0.002,
0.000,0.000,0.000,1.000

T_C1toI:
-1.000,-0.003,0.010,-0.071,
0.003,-1.000,0.004,-0.010,
0.010,0.004,1.000,0.003,
0.000,0.000,0.000,1.000

TIME: 55.563 seconds

================================================================================REQUIRED process [run_serial_msckf-2] has died!
process has finished cleanly
log file: /home/olorin/.ros/log/e093a38c-b726-11ea-852f-18dbf203f0de/run_serial_msckf-2*.log
Initiating shutdown!
================================================================================
[run_serial_msckf-2] killing on exit
`

Why am I not getting any visualization

goldbattle, 2020-06-25:
We provide a rviz configuration file.
If you open that it should visualize.
Check out the tutorial on the website and make sure that works for you also.

goldbattle, 2020-06-26:
You can open the `display.rviz` in here with ROS rviz:
https://github.com/rpng/open_vins/tree/master/ov_msckf/launch

ShrutheeshIR, 2020-06-26:
Thanks a ton!

closed 2020-06-26 by ShrutheeshIR