rpng/open_vins #77: system is divergent after the hardware(vio-sensors) stay static for a while?

Wangxuefeng92, 2020-06-28:
I carried my vio-sensor, and  walked around  for a long-term, then I stayed  static for a while. then I found that the trajectory is divergent, also the residual is divergent...

goldbattle, 2020-06-28:
This can happen if you lose track of your SLAM features and there isn't a zero velocity update. We will hopefully open source the zero velocity update soon (not sure on the timeline). But you can avoid this by making sure you track a lot of SLAM features and they are able to be continuously tracked when you are still.

The underlying computer vision issue is that you cannot triangulate any *new* features if you are still since there is no parallax between features.

Wangxuefeng92, 2020-06-29:
thanks for your reply！I am also wandering that how to detect the “STATIC” condition for a vio system. Because the vision or imu measurement is so noisy, and the vio system can also slight jitter sometimes.

rpng-guest, 2020-07-08:
PR #79 should address this, feel free to try it out (does require some tuning of the detection thresholds).

closed 2020-07-08 by rpng-guest