rpng/open_vins #78: Fly soon after initializing on TUM Visual Inertial Dataset

highlightz, 2020-07-02:
- Dataset：dataset-room1_512_16.bag
- config file：
```
<launch>

    <!-- mono or stereo and what ros bag to play -->
    <arg name="max_cameras" default="2" />
    <arg name="bag"         default="/home/xs/dataset/tumvi/dataset-room1_512_16.bag" />

    <!-- saving trajectory paths -->
    <arg name="dosave"      default="false" />
    <arg name="path_est"    default="/tmp/traj_estimate.txt" />

    <!-- MASTER NODE! -->
    <node name="run_subscribe_msckf" pkg="ov_msckf" type="run_subscribe_msckf" output="screen" clear_params="true" required="true">

        <!-- bag topics -->
        <param name="topic_imu"      type="string" value="/imu0" />
        <param name="topic_camera0"  type="string" value="/cam0/image_raw" />
        <param name="topic_camera1"  type="string" value="/cam1/image_raw" />

        <!-- world/filter parameters -->
        <param name="use_fej"                type="bool"   value="true" />
        <param name="use_imuavg"             type="bool"   value="true" />
        <param name="calib_cam_extrinsics"   type="bool"   value="true" />
        <param name="calib_cam_intrinsics"   type="bool"   value="true" />
        <param name="calib_cam_timeoffset"   type="bool"   value="true" />
        <param name="calib_camimu_dt"        type="double" value="0.0" />
        <param name="max_clones"             type="int"    value="15" />
        <param name="max_slam"               type="int"    value="30" />
        <param name="max_cameras"            type="int"    value="$(arg max_cameras)" />
        <param name="dt_slam_delay"          type="double" value="5" />
        <param name="feat_representation"    type="string" value="ANCHORED_3D" />
        <param name="init_window_time"       type="double" value="1.0" />
        <param name="init_imu_thresh"        type="double" value="0.8" /> <!-- works for 1-5 -->
        <!-- <param name="init_imu_thresh"        type="double" value="0.5" /> &lt;!&ndash; works for 6 &ndash;&gt;-->
        <rosparam param="gravity">[0.0,0.0,9.80766]</rosparam>

        <!-- tracker/extractor properties -->
        <param name="use_klt"          type="bool"   value="true" />
        <param name="num_pts"          type="int"    value="200" />
        <param name="fast_threshold"   type="int"    value="10" />
        <param name="grid_x"           type="int"    value="5" />
        <param name="grid_y"           type="int"    value="3" />
        <param name="min_px_dist"      type="int"    value="5" />
        <param name="knn_ratio"        type="double" value="0.65" />

        <!-- aruco tag/mapping properties -->
        <param name="use_aruco"        type="bool"   value="false" />
        <param name="num_aruco"        type="int"    value="1024" />
        <param name="downsize_aruco"   type="bool"   value="true" />

        <!-- sensor noise values / update -->
        <param name="up_msckf_sigma_px"            type="double"   value="1" />
        <param name="up_msckf_chi2_multipler"      type="double"   value="1" />
        <param name="up_slam_sigma_px"             type="double"   value="1" />
        <param name="up_slam_chi2_multipler"       type="double"   value="1" />
        <param name="up_aruco_sigma_px"            type="double"   value="1" />
        <param name="up_aruco_chi2_multipler"      type="double"   value="1" />
        <param name="gyroscope_noise_density"      type="double"   value="0.00016" />
        <param name="gyroscope_random_walk"        type="double"   value="0.000022" />
        <param name="accelerometer_noise_density"  type="double"   value="0.0028" />
        <param name="accelerometer_random_walk"    type="double"   value="0.00086" />

        <!-- camera intrinsics -->
        <param name="cam0_is_fisheye" type="bool" value="true" />
        <param name="cam1_is_fisheye" type="bool" value="true" />
        <rosparam param="cam0_k">[190.97847715128717, 190.9733070521226, 254.93170605935475, 256.8974428996504]</rosparam>
        <rosparam param="cam0_d">[0.0034823894022493434, 0.0007150348452162257, -0.0020532361418706202, 0.00020293673591811182]</rosparam>
        <rosparam param="cam1_k">[190.44236969414825, 190.4344384721956, 252.59949716835982, 254.91723064636983]</rosparam>
        <rosparam param="cam1_d">[0.0034003170790442797, 0.001766278153469831, -0.00266312569781606, 0.0003299517423931039]</rosparam>

        <!-- camera extrinsics -->
        <rosparam param="T_C0toI">
            [
            -0.99952504,  0.00750192, -0.02989013,  0.04557484,
             0.02961534, -0.03439736, -0.99896935, -0.07116180,
            -0.00852233, -0.99938008,  0.03415885, -0.04468125,
            0.0, 0.0, 0.0, 1.0
            ]
        </rosparam>
        <rosparam param="T_C1toI">
            [
            -0.99951105,  0.00810408, -0.03019914, -0.05545634,
             0.03029912,  0.01251164, -0.99946257, -0.06925002,
            -0.00772188, -0.99988889, -0.01275107, -0.04745286,
            0.0, 0.0, 0.0, 1.0
            ]
        </rosparam>

    </node>

    <!-- play the dataset -->
    <node pkg="rosbag" type="play" name="rosbag" args="-d 1 $(arg bag)" required="true"/>

    <!-- record the trajectory if enabled -->
    <group if="$(arg dosave)">
        <node name="recorder_estimate" pkg="ov_eval" type="pose_to_file" output="screen">
            <param name="topic"      type="str" value="/ov_msckf/poseimu" />
            <param name="topic_type" type="str" value="PoseWithCovarianceStamped" />
            <param name="output"     type="str" value="$(arg path_est)" />
        </node>
    </group>

</launch>
```

- performance:
[image]

- other info:
  * I run the same program in monocular mode(setting max_cameras as 1), it still does not behaves correctly
  * I run the same program in both monocular and stereo mode on EuRoC datasets(V2_03_difficult), it behaves correctly
 
Does above mentioned details cue that fisheye cameras(as TUM V-I Datasets) are not supported by your program?

goldbattle, 2020-07-02:
The issue with the TUM datasets is that they do not start from zero
velocity. You mind need to play around with the IMU init threshold and bag
start time. I recommend checking out the dataset documentation page for
further details:
https://docs.openvins.com/gs-datasets.html#gs-data-tumvi

We do support fisheye cameras and calibrating them online, and there is an
example launch in the repo for the TUM-VI. Let me know if you have any more
questions or can't get it to work.

goldbattle, 2020-07-02:
You also should be able to download the ROS bags directly from their website as compared to making them yourself from a zip file.

goldbattle, 2020-07-09:
Feel free to re-open if you run into the issue again. Thanks!

closed 2020-07-09 by goldbattle

highlightz, 2020-07-14:
I download ROS bag dataset-room1_512_16.bag from TUM official website and run with the above lanch file again, it works correctly.  Below is the trajectory in rviz, and it looks smooth and with small drift when coming back to starting point:

[image]

Also, I change parameter init_imu_thresh from 0.5 to 0.7(your recommended value in run_ros_tumvi.sh) and run it again, it also works well.  
So I think there is something wrong when I make bag file from zip file though I don't know why.

In fact, the file size is so different, the official size of dataset-room1_512_16.bag is 3.0G, but the one I make from zip is only 1.5G. I will check the reason myself.

highlightz, 2020-07-14:
Below is a comparison of my self-made bag and tumvi official bag(first is mine, second is tum official):

```
xs@ubuntu:~/dataset/tumvi$ rosbag info dataset-room1_512_16_my.bag 
path:        dataset-room1_512_16_my.bag
version:     2.0
duration:    2:20s (140s)
start:       Mar 09 2018 01:31:48.20 (1520530308.20)
end:         Mar 09 2018 01:34:08.23 (1520530448.23)
size:        1.4 GB
messages:    33764
compression: none [1894/1894 chunks]
types:       sensor_msgs/Image [060021388200f6f0f447d0fcd9c64743]
             sensor_msgs/Imu   [6a62c6daae103f4ff57a132d6f95cec2]
topics:      /cam0/image_raw    2821 msgs    : sensor_msgs/Image
             /cam1/image_raw    2821 msgs    : sensor_msgs/Image
             /imu0             28122 msgs    : sensor_msgs/Imu

xs@ubuntu:~/dataset/tumvi$ rosbag info dataset-room1_512_16.bag 
path:        dataset-room1_512_16.bag
version:     2.0
duration:    2:21s (141s)
start:       Mar 09 2018 01:31:48.18 (1520530308.18)
end:         Mar 09 2018 01:34:09.23 (1520530449.23)
size:        2.8 GB
messages:    50305
compression: none [57/57 chunks]
types:       geometry_msgs/TransformStamped [b5764a33bfeb3588febc2682852579b0]
             sensor_msgs/Image              [060021388200f6f0f447d0fcd9c64743]
             sensor_msgs/Imu                [6a62c6daae103f4ff57a132d6f95cec2]
topics:      /cam0/image_raw               2821 msgs    : sensor_msgs/Image             
             /cam1/image_raw               2821 msgs    : sensor_msgs/Image             
             /imu0                        28122 msgs    : sensor_msgs/Imu               
             /vrpn_client/raw_transform   16541 msgs    : geometry_msgs/TransformStamped
```

I can not yet figure out the difference in image, imu or their timestamps.