rpng/open_vins #83: Trajectory drifts away.

zgxsin, 2020-07-09:
I am running OpenVins  based pgeneva_ros_eth.launch using my own datasets. I changed the parameters inside that launch file, such as camera intrinsic and extrinsic and rosbag path. The dateset is obtained from a synchronized monocular camera and IMU.
However, the trajectory just drifts away:
```
q_GtoI = -0.124,0.149,-0.706,0.682 | p_IinG = 102.369,24.533,-15.013 | dist = 107.64 (meters)
bg = 0.0224,-0.0269,0.0188 | ba = 0.0563,0.0260,0.0765
camera-imu timeoffset = -0.01609
cam0 intrinsics = 625.477,627.345,625.626,506.976 | -0.009,0.015,-0.016,0.006
cam0 extrinsics = -0.001,0.707,-0.707,0.000 | -0.007,-0.009,-0.254
[TIME]: 0.0057 seconds for tracking
[TIME]: 0.0002 seconds for propagation
[TIME]: 0.0001 seconds for MSCKF update (0 features)
[TIME]: 0.0000 seconds for SLAM update (0 feats)
[TIME]: 0.0002 seconds for SLAM delayed init (0 feats)
[TIME]: 0.0015 seconds for marginalization (11 clones in state)
[TIME]: 0.0078 seconds for total
q_GtoI = -0.122,0.151,-0.706,0.681 | p_IinG = 103.358,24.640,-15.180 | dist = 108.65 (meters)
bg = 0.0224,-0.0269,0.0188 | ba = 0.0563,0.0260,0.0765
camera-imu timeoffset = -0.01609
cam0 intrinsics = 625.477,627.345,625.626,506.976 | -0.009,0.015,-0.016,0.006
cam0 extrinsics = -0.001,0.707,-0.707,0.000 | -0.007,-0.009,-0.254
[TIME]: 0.0067 seconds for tracking
[TIME]: 0.0002 seconds for propagation
[TIME]: 0.0001 seconds for MSCKF update (0 features)
[TIME]: 0.0000 seconds for SLAM update (0 feats)
[TIME]: 0.0002 seconds for SLAM delayed init (0 feats)
[TIME]: 0.0010 seconds for marginalization (11 clones in state)
[TIME]: 0.0081 seconds for total
q_GtoI = -0.121,0.153,-0.706,0.681 | p_IinG = 104.344,24.745,-15.347 | dist = 109.65 (meters)
bg = 0.0224,-0.0269,0.0188 | ba = 0.0563,0.0260,0.0765
camera-imu timeoffset = -0.01609
cam0 intrinsics = 625.477,627.345,625.626,506.976 | -0.009,0.015,-0.016,0.006
cam0 extrinsics = -0.001,0.707,-0.707,0.000 | -0.007,-0.009,-0.254
[TIME]: 0.0055 seconds for tracking
[TIME]: 0.0002 seconds for propagation
[TIME]: 0.0001 seconds for MSCKF update (0 features)
[TIME]: 0.0000 seconds for SLAM update (0 feats)
[TIME]: 0.0003 seconds for SLAM delayed init (0 feats)
[TIME]: 0.0010 seconds for marginalization (11 clones in state)
[TIME]: 0.0071 seconds for total
q_GtoI = -0.120,0.154,-0.706,0.681 | p_IinG = 105.340,24.849,-15.515 | dist = 110.67 (meters)
bg = 0.0224,-0.0269,0.0188 | ba = 0.0563,0.0260,0.0765
camera-imu timeoffset = -0.01609
cam0 intrinsics = 625.477,627.345,625.626,506.976 | -0.009,0.015,-0.016,0.006
cam0 extrinsics = -0.001,0.707,-0.707,0.000 | -0.007,-0.009,-0.254
[TIME]: 0.0053 seconds for tracking
[TIME]: 0.0002 seconds for propagation
[TIME]: 0.0001 seconds for MSCKF update (0 features)
[TIME]: 0.0000 seconds for SLAM update (0 feats)
[TIME]: 0.0001 seconds for SLAM delayed init (0 feats)
[TIME]: 0.0010 seconds for marginalization (11 clones in state)
[TIME]: 0.0066 seconds for total
q_GtoI = -0.119,0.156,-0.706,0.680 | p_IinG = 106.342,24.952,-15.685 | dist = 111.69 (meters)
bg = 0.0224,-0.0269,0.0188 | ba = 0.0563,0.0260,0.0765
camera-imu timeoffset = -0.01609
cam0 intrinsics = 625.477,627.345,625.626,506.976 | -0.009,0.015,-0.016,0.006
cam0 extrinsics = -0.001,0.707,-0.707,0.000 | -0.007,-0.009,-0.254
[TIME]: 0.0056 seconds for tracking
[TIME]: 0.0002 seconds for propagation
[TIME]: 0.0001 seconds for MSCKF update (0 features)
[TIME]: 0.0000 seconds for SLAM update (0 feats)
[TIME]: 0.0002 seconds for SLAM delayed init (0 feats)
[TIME]: 0.0010 seconds for marginalization (11 clones in state)
[TIME]: 0.0071 seconds for total
q_GtoI = -0.118,0.157,-0.706,0.680 | p_IinG = 107.347,25.053,-15.855 | dist = 112.71 (meters)
bg = 0.0224,-0.0269,0.0188 | ba = 0.0563,0.0260,0.0765
camera-imu timeoffset = -0.01609
cam0 intrinsics = 625.477,627.345,625.626,506.976 | -0.009,0.015,-0.016,0.006
cam0 extrinsics = -0.001,0.707,-0.707,0.000 | -0.007,-0.009,-0.254
[TIME]: 0.0056 seconds for tracking
[TIME]: 0.0002 seconds for propagation
[... 81 lines cut ...]
cam0 extrinsics = -0.001,0.707,-0.707,0.000 | -0.007,-0.009,-0.254
[TIME]: 0.0051 seconds for tracking
[TIME]: 0.0002 seconds for propagation
[TIME]: 0.0001 seconds for MSCKF update (0 features)
[TIME]: 0.0000 seconds for SLAM update (0 feats)
[TIME]: 0.0002 seconds for SLAM delayed init (0 feats)
[TIME]: 0.0009 seconds for marginalization (11 clones in state)
[TIME]: 0.0065 seconds for total
q_GtoI = -0.125,0.156,-0.703,0.683 | p_IinG = 115.614,25.827,-17.260 | dist = 121.13 (meters)
bg = 0.0224,-0.0269,0.0188 | ba = 0.0563,0.0260,0.0765
camera-imu timeoffset = -0.01609
cam0 intrinsics = 625.477,627.345,625.626,506.976 | -0.009,0.015,-0.016,0.006
cam0 extrinsics = -0.001,0.707,-0.707,0.000 | -0.007,-0.009,-0.254

```

This part is not updated after some time:
```
bg = 0.0224,-0.0269,0.0188 | ba = 0.0563,0.0260,0.0765
camera-imu timeoffset = -0.01609
cam0 intrinsics = 625.477,627.345,625.626,506.976 | -0.009,0.015,-0.016,0.006
cam0 extrinsics = -0.001,0.707,-0.707,0.000 | -0.007,-0.009,-0.254
```

Does anyone have any idea?

goldbattle, 2020-07-09:
It looks like you have to features tracks being successfully triangulated.
Is your camera intrinsics correct? Your camera centers are pretty large.
Order is fx, fy, cx, cy.

zgxsin, 2020-07-09:
Hi, thanks for your reply. Yes, i am using the right order `fx, fy, cx, cy` as well as the right intrinsics provided by Kalibr.

zgxsin, 2020-07-09:
I just found the the covariance of my IMU reading is zero. Maybe that is related.

zgxsin, 2020-07-16:
I encountered this issue using simulation data. There is no issue if use the real-world data collected on the same platform. I guess there is something wrong with my simulation data. But openvins is not stable on my real-world data. It won't drift away in most time. But it still drifts away sometimes. How can i make it robsut?

goldbattle, 2020-07-20:
There might be issues with doing zero IMU noise (for sure in the realworld, but might work ok in simulation).

If you can pose your trajectory file we can also try to debug that.
It might be there is a place where the unit stops, or your simulation launch file isn't setup correctly.
My recommendation is to use the example simulation launch file from the repo, and just use a different pose file.

You can checkout the different simulation trajectories we have used to see if they work ok for you:
https://github.com/rpng/open_vins/tree/master/ov_data/sim

zgxsin, 2020-08-06:
Will close this issue since it is related to bad simulation data generated by our own.

closed 2020-08-06 by zgxsin