rpng/open_vins #84: mathematical derivation of "Qd".

suho0515, 2020-07-14:
hello sir,
i got a question while looking "predict_and_compute()" function.

the codes are below.

```
// Construct our discrete noise covariance matrix
// Note that we need to convert our continuous time noises to discrete
// Equations (129) amd (130) of Trawny tech report
Eigen::Matrix<double,12,12> Qc = Eigen::Matrix<double,12,12>::Zero();
Qc.block(0,0,3,3) = _noises.sigma_w_2/dt*Eigen::Matrix<double,3,3>::Identity();
Qc.block(3,3,3,3) = _noises.sigma_a_2/dt*Eigen::Matrix<double,3,3>::Identity();
Qc.block(6,6,3,3) = _noises.sigma_wb_2/dt*Eigen::Matrix<double,3,3>::Identity();
Qc.block(9,9,3,3) = _noises.sigma_ab_2/dt*Eigen::Matrix<double,3,3>::Identity();

// Compute the noise injected into the state over the interval
Qd = G*Qc*G.transpose();
Qd = 0.5*(Qd+Qd.transpose());

```

i can see that the continous-time noise covariance matrix "Qc" is made.
(picture below is about just gyro, so i can see that when we consider gyro and acc, the matrix would be "12x12")

[image]

and make "Qc" to discrete-time noise matrix "Qd" using

[image]

and we multiply "Qd" with "G" for next work.

[image]

so... i can understand till here

```
// Construct our discrete noise covariance matrix
// Note that we need to convert our continuous time noises to discrete
// Equations (129) amd (130) of Trawny tech report
Eigen::Matrix<double,12,12> Qc = Eigen::Matrix<double,12,12>::Zero();
Qc.block(0,0,3,3) = _noises.sigma_w_2/dt*Eigen::Matrix<double,3,3>::Identity();
Qc.block(3,3,3,3) = _noises.sigma_a_2/dt*Eigen::Matrix<double,3,3>::Identity();
Qc.block(6,6,3,3) = _noises.sigma_wb_2/dt*Eigen::Matrix<double,3,3>::Identity();
Qc.block(9,9,3,3) = _noises.sigma_ab_2/dt*Eigen::Matrix<double,3,3>::Identity();

// Compute the noise injected into the state over the interval
Qd = G*Qc*G.transpose();
```

but why should we add "Qd" to "Qd.tanspose()" and multiply with 0.5?

`Qd = 0.5*(Qd+Qd.transpose());
`

I've looked for Docs, Paper but couldn't find clear answer.
would you let me know that?

anyway, thank you for your amazing works again. :)

goldbattle, 2020-07-14:
That operation probably isn't needed, but it is to enforce that the matrix
is symmetrical.

suho0515, 2020-07-15:
> That operation probably isn't needed, but it is to enforce that the matrix is symmetrical.
> […](#)

thanks for the answer,

i'll close this issue then!

closed 2020-07-15 by suho0515