rpng/open_vins #86: the memory occupied by the program is getting larger and larger

Teemoshan, 2020-07-21:
[image]
Hi, I encountered some problems in the process of using, the memory occupied by the program is getting larger and larger, as shown in the picture, MSCKF-VIO does not have this problem.

goldbattle, 2020-07-21:
Does this only happen when you run serial, or also the ros subscriber node?

Teemoshan, 2020-07-21:
[image]
the ros subscriber node  also  takes up a lot of memory

Teemoshan, 2020-07-22:
When max_cameras is 1, the program is normal. But when I use two camera , the memory occupied by the program is getting larger. This also happens on the euroc dataset, you can also try.

goldbattle, 2020-07-22:
Yeah this isn't something that should be happening. Maybe something to do with visualization or how multicam tracks are handled. I will try to take a look and see if I can see anything.

Teemoshan, 2020-09-02:
Have you solved this problem？ @goldbattle

goldbattle, 2020-09-05:
I have not due to time constraints, hopefully I will soon, but if you have any further investigations feel free to post, thanks!

kimyoungji, 2020-09-26:
@goldbattle

I found that this is caused by line 151-159 in 'TrackKLT::feed_stereo'.
This code is about building optical flow pyramids using cv functions and threads. 
I don't know the exact reasons why using threads arouses memory leak, it was fine when I revise it as
    cv::buildOpticalFlowPyramid(img_left, imgpyr_left, win_size, pyr_levels);
    cv::buildOpticalFlowPyramid(img_right, imgpyr_right, win_size, pyr_levels);
I hope this will be helpful for your upcoming updates.
Thanks!

goldbattle, 2020-09-30:
Hi, I wasn't able to reproduce this on my side.
What version of OpenCV are you using @Teemoshan and @kimyoungji ?

kimyoungji, 2020-10-01:
Mine is opencv 3.3.1.

goldbattle, 2020-12-01:
This should be addressed in #117, valgrind doesn't report any memory leaks and we are switched over to smart points.
This might be specific to an OpenCV version, try upgrading if it is still an issue or use your above fix.

closed 2020-12-01 by goldbattle