rpng/open_vins #87: How to use dataset of kaist, I meet error when I load data by fileplayer

sjlszz, 2020-07-21:


goldbattle, 2020-07-22:
Can you provide a bit more details of the exact problem you are having? Thanks.

sjlszz, 2020-07-22:
I follow the instruction from the website of KAIST dataset, I try to use the Open_vins and I load Urban28 , when I run rosrun file_player file_player, the load path is wrong.And the structure of dataset is same as the website of KAIST dataset.I only need image and IMU.

goldbattle, 2020-07-22:
Their player requires all data to be downloaded. There is a load button
from the GUI to open the folder where the IMU, lidar, vehicle info etc. is.

If you could post the exact error, or screenshot it, that would be great. I
recommend you check out their repo on how to use it:
https://github.com/irapkaist/file_player#4-load-data-files-and-play

sjlszz, 2020-07-23:
ok,i will try it

sjlszz, 2020-07-23:
erminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 18446744073709551615) >= this->size() (which is 2001)

when I use kaist ,  I meet the problem above,what's matter?

sjlszz, 2020-07-24:
only certain topics, I need.

goldbattle, 2020-08-28:
This is due to zero velocity update IMU selection logic, will track this in Issue #90.

closed 2020-08-28 by goldbattle