rpng/open_vins #88: How are Euroc ground truths obtained in ov_data

StevePhan101, 2020-07-23:
Hi, thanks for open-sourcing a useful VIO tool.

I'm evaluating the estimated trajectory of Euroc V1_01_easy. I saw a ground truth file V1_01_easy.csv/txt files in ov_data/euroc_mav. When I checked with the ground truth file in the Euroc souce: http://robotics.ethz.ch/~asl-datasets/ijrr_euroc_mav_dataset/vicon_room1/V1_01_easy/V1_01_easy.bag. The ground truth is stored in topic: /vicon/firefly_sbx/firefly_sbx. I can see that the source ground truth and open_vins ground truth are different.

When I evaluate the estimated trajectory path_est.txt with the open_vins ground truth, the performance is solid. However, the trajectory drifts away from the source ground truth.

Output for evaluating on open_vins gt:
```
[COMP]: 2895 poses in V1_01_easy => length of 58.35 meters
[TRAJ]: q_ESTtoGT = -0.006, -0.004, -0.143, 0.990 | p_ESTinGT = 0.866, 2.179, 0.940 | s = 1.00
[COMP]: 2680 poses in 1595400591_estimate => length of 57.11 meters
```

[image]
[image]

Output for evaluating on source gt:
```
[COMP]: 14629 poses in V1_01_gt => length of 60.12 meters
[TRAJ]: q_ESTtoGT = 0.076, -0.535, 0.059, 0.839 | p_ESTinGT = 0.770, 2.178, 1.073 | s = 1.00
[COMP]: 2680 poses in 1595400591_estimate => length of 57.11 meters
```
[image]
[image]

goldbattle, 2020-07-23:
I believe that topic is of the vicon marker MAV frame, which does not
coincide with the IMU frame that most estimation methods estimate.
OpenVINS publishes the IMU sensor frame.

The csv files for groundtruth are pulled from the ASL zip files (should be
a groundtruth folder in it).
This is then converted using the ov_eval format_converter utility (this is
the text file in the repo).

I am surprised with the large error you have shown in the first figure,
what method are you using to align?
Typically we use  posyaw  to align it to find the ATE / RMSE of the
trajectory.
This corresponds to the 4 unobservable DOF that VINS has and are
arbitrarily chosen on startup.

StevePhan101, 2020-07-23:
Thanks for the quick response.

I was using se3single method for evaluation.
I have converted the data.csv from the source to txt using format_converter.cpp.

Yes, you are right, using posyaw yields better result. Here is the output when using posyaw evaluation,

Evaluation on source_gt.txt:
[image]

Evaluation on openvins_gt.txt:
[image]

I can notice that the ov_gt.txt and src_gt.txt are slightly different in terms of both timestamp and pose values. Even though the difference is insignificant, I'm still curious why the difference occurs. Additionally, I also see the files V1_01_easy_original and V1_01_easy and they are different as well. Could you please give me some clarifications?

Thank you in advance.

goldbattle, 2020-07-23:
That difference should be described here on the dataset page:
https://docs.openvins.com/gs-datasets.html#gs-data-euroc

StevePhan101, 2020-07-23:
Thanks for your information.

closed 2020-07-23 by StevePhan101

goldbattle, 2021-06-09:
For those that re-visit this issue, please see the vicon2gt repository if you want to generate this yourself:
https://github.com/rpng/vicon2gt

habib-Boloorchi, 2022-03-17:
Hi Do you know if the data in ground truth is the distance from origin or the distance from last timestamp.