rpng/open_vins #91: Problems replicating EuRoC results, pgeneva_ros_eth.launch produces timestamps inconsistent with ground truth.

ab-10, 2020-08-11:
I'm trying to replicate the results of your paper [OpenVINS:  A  Research  Platform  for  Visual-Inertial  Estimation](http://udel.edu/~pgeneva/downloads/papers/c10.pdf), using `pgeneva_ros_eth.launch` file.

I'm using EuRoC V1_02_medium.

I downloaded the bag file and ASL dataset from here: https://projects.asl.ethz.ch/datasets/doku.php?id=kmavvisualinertialdatasets

I used `data.csv` as the ground-truth (located in `V1_02_medium/mav0/state_groundtruth_estimate0`, where `V1_02_medium` is the unziped ASL dataset).

I converted `data.csv` to `data.txt` using `rosrun ov_eval format_converter data.csv`

In `ov_msckf/launch/pgeneva_ros_eth.launch` I have changed the `bag` arg to my system path to `V1_02_medium.bag` and I've switched `dosave` to true.

However, the timestamps of the newly created path estimate file (in the location specified by arg `path_est`) do not match the timestamps of `data.txt` file.

`data.txt` starts at timestamp (s) `1403715524.90714` and ends at `1403715608.41214`.

Path estimate produced by `pgeneva_ros_eth.launch` starts at `1403715278.16214` and ends in `1403715418.71220`.

Consequently, the path estimate ends before the ground truth data begins.

If there's a more straightforward way to replicate the results, please let me know.

goldbattle, 2020-08-12:
Since we estimate the IMU camera time offset they shouldn't line up. The ground truth is in the IMU clock frame, thus we append the current camera IMU time offset when we publish the poses. If you disable the toff it should line up if you use a time offset of zero (I have not tested, but logically it should be the case). 

Let me know if this answers your question.

ab-10, 2020-08-12:
> If you disable the toff it should line up if you use a time offset of zero

The error persists when `bag_start` arg is set to `0`, is this what you meant?

I should also specify, that the error only occurs with EuRoC `V1_02_medium`, I've tested with the other EuRoC bag files and they provide a sufficient number of overlapping timestamps with the ground truth to evaluate using `error_singlerun` script (which is not the case when running `V1_02_medium`).

goldbattle, 2020-08-12:
Can you compare to the CSV files here:
https://www.github.com/rpng/open_vins/tree/master/ov_data%2Feuroc_mav

These should be the ones from the zip ASL files, besides the V101 entry.

ab-10, 2020-08-12:
The V1_02 ground truth entry I used is identical to the one provided in https://www.github.com/rpng/open_vins/tree/master/ov_data%2Feuroc_mav

goldbattle, 2020-08-13:
The ones that I use are from the ov_data folder, so I am not sure why there is a difference.
Are you by chance using sim_time or some other sort of parameter to change how ROS time works?
From rosbag info I get that the bag:
```
path:        V1_02_medium.bag
version:     2.0
duration:    1:28s (88s)
start:       Jun 25 2014 12:58:43.93 (1403715523.93)
end:         Jun 25 2014 13:00:12.05 (1403715612.05)
```

When I run it I get that the data that is saved to the traj_estimate.txt:
- Starts at `1403715528.46214` (starts 4.5 seconds after the first bad timestamp)
- Ends at `1403715609.31193` (ends 2.73 seconds before the bag ends)

I am running the following command to look at the error for a single run (default launch file):
```
rosrun ov_eval error_singlerun posyaw /tmp/traj_estimate.txt src/open_vins/ov_data/euroc_mav/V1_02_medium.txt
```

goldbattle, 2020-08-28:
Reopen if you run into the issue again.

closed 2020-08-28 by goldbattle