rpng/open_vins #92: Question About fast_state_propagate()

JeffreyZh4ng, 2020-08-13:
We saw in the V2 release that there was this function fast_state_propagate which will provide higher frequency pose estimations between updates. We want to use this function to do a pose prediction but based on our understanding of this function, it only operates over available IMU readings, and doesn't predict into the future.

To get around this, we want to append spoofed IMU readings and let fast_state_propagate predict to the point in time we want (which will always be past any real IMU readings currently recorded).

Can you provide some insight into if this is a good approach for what we want to do? Thanks!

goldbattle, 2020-08-13:
The first think that comes to mind is to use the published angular and linear velocity to linearly interpolate into the future to get a pose. This of course breaks down if the extrapolation goes to far. Another way would be to fit a spline/polynomial to the current pose history (maybe last 3-5 poses) and from that evaluate into the future. I would recommend you try just linearly interpolating into the future which wouldn't require any editing of the current code, and could just be on the "client side" of things.

[image]
[image]

In this case the [published angular velocity](https://github.com/rpng/open_vins/blob/master/ov_msckf/src/state/Propagator.cpp#L176-L177) should have bias already removed. You could instead use the exp of SO(3) and do a rotation matrix, I have just shown quaternion as this is exactly what we do during propagation.

JeffreyZh4ng, 2020-08-13:
Do you know anything specifically about the way we're approaching it (using the last IMU measurement to spoof IMU values up to a point in the future, then passing in these appended spoofed values to fast_state_propagate)? We're creating these fake IMU values using a constant acceleration model. 

Also can you confirm that fast_state_propagate does not change (or completely restores) the state passed into the function? It seems like we're getting some very inconsistent pose fly away issues after using the method I mentioned

goldbattle, 2020-08-14:
I have personally never tried that way, and not sure if that would work.
My recommendation is to do the following (untested code of course and you need to pass in timestamp_future also):
```
double dt = timestamp_future-timestamp;
Eigen::Vector3d delta_theta = state_plus.block(10,0,3,1) * dt;
Eigen::Vector3d delta_pos = state_plus.block(7,0,3,1) * dt;
Eigen::Matrix3d R_future = ov_core::exp_so3(-delta_theta)*state_plus.block(0,0,4,1);
Eigen::Matrix3d p_future = state_plus.block(4,0,3,1) + delta_pos;
```