rpng/open_vins #94: does local perturbation parameterization and fej in evaluate jacobians can hold the right  nullspace of the observability matrix?

tadashi-hamada196, 2020-08-24:
in paper High-precision, consistent ekf-based visual-inertial odometry，it said using global perturbation and fej can keep right nullspace of the observability matrix。but in open_vins，I see it use  local perturbation parameterization，I become confuse，can you explain this?

goldbattle, 2020-08-28:
We address the observability issue through using First-Estimate Jacobians. This basically means that if we multiplied the state transition matrices together from sequential timesteps they should be valid. There is a bit more detail explained here:
https://docs.openvins.com/fej.html#fej-fej

By using the local orientation / local perturbation, our state transition does not have the identity for rotation and instead has a rotation. We could change everything to right hand such as in the MSCKF 2.0 paper, but since we are already using FEJ there shouldn't be a big difference, and we would stray a bit too far from conventional MSCKF literature.

Off the top of my head, (besides the paper in the above docs), this paper can also provide a starting point for your readings:
http://people.csail.mit.edu/ghuang/paper/Huang2010IJRR.pdf

closed 2020-09-07 by goldbattle

carolo44, 2022-05-31:
have you ever tried to use the global orientation perturbation instead of the local one? any experimental result?

goldbattle, 2022-06-02:
Not off the top of my head, but I don't think there is that impactful gain
in the realworld.
You can give it a try and compare through simulation if you are curious.