rpng/open_vins #97: T265 calibration using Kalibr

JinyongJeong, 2020-08-30:
Hi. 

I am using T265 for openVins.

I calibrated T265 using Kalibr. 

I found launch file for T265 in `issues` list. 

In this launch file, intrinsic of T265 camera was 

        <rosparam param="cam0_k">[287.5222548531953, 287.7759196963578, 419.94592910576455, 394.2977562606552]</rosparam>
        <rosparam param="cam0_d">[-0.008152232448426595, 0.06139283207665097, -0.06682430814303411, 0.01756431204855009]</rosparam>
        <rosparam param="cam1_k">[287.2648972756499, 287.5526675605301, 417.53890746140695, 399.1202543659908]</rosparam>
        <rosparam param="cam1_d">[-0.006601389878082893, 0.06285624670204437, -0.06624718433022161, 0.016178100033794867]</rosparam>

However, when I calibrated using Kalibr. The result was 

        <rosparam param="cam0_k">[564.1673377333339,559.4539715388482,418.33652976109613,413.2744935489872]</rosparam>
        <rosparam param="cam0_d">[-0.2558938171490923,-0.0460934839068549,-0.00414934653924308,0.0007768208763751949]</rosparam>
        <rosparam param="cam1_k">[556.9637359565154,553.816051011247,419.47350438648533,406.59047995751183]</rosparam>
        <rosparam param="cam1_d">[-0.2204827920930757,-0.08227157719154526,-0.00483280276623621,-0.0037763597533052246]</rosparam>

Intrinsic value of my result is much bigger than yours. 
I use rad-tan model for calibration (calibration was failed when I use equ-distance model).

Can you kindly let me know, how to calibrate well for using T265 from scratch?

goldbattle, 2020-08-31:
This guide on the doc website is exactly what we normally due to calibrate any of our sensors:
https://docs.openvins.com/gs-calibration.html

Maybe you have your board yaml setup incorrectly if you are not using the default size?

JinyongJeong, 2020-08-31:
@goldbattle Did you use rad-tan model for T265 camera? I printed calibration board in A4 side and changed the yaml file with real size of printed marker. 
When I use pinhole-eque model, it couldn't converge to final calibrated value.

yangyulin, 2020-08-31:
Hi,
I recently just tried T265 with openvins.
It is better to use the equi_dist model to calibrate the t265 with kalibr.

The results you presented with Kalibr seem like a rad-tan model.

When I did the calibration, I calibrated several times. A good(dynamic and
random) motion is important for the calibration.  Sometimes it cannot
converge, but sometimes it did.
If you check the output on the screen of Kalibr, the dx change actually is
very small after a few iterations. You can set the max iteration times
(like 10) to save your time. You can also set lower camera frequency to
speed up the calibration.
The following is my calibration results:
[image: image.png]

And you can see the trajectory from the calibration bag with open-vins
using the calibration results;

closed 2020-09-15 by goldbattle