rpng/open_vins #98: Single Camera Intrinsic Calibration

SubMishMar, 2020-09-09:
Hi,
I am following the tutorial given in https://docs.openvins.com/gs-calibration.html for calibrating my camera imu system. In the 1st step I am trying to calibrate the intrinsics of the single camera I am using. As described in the tutorial I collect images of april tag at low frequency. But when I try to calibrate using kalibr it says that calibration has failed because it cannot detect more than one camera. Although I realize that kalibr is a multi camera calibration system, is there a way to just calibrate a single camera?
Adding some screen shots for your perusal.
[
[image]
[image]

](url)

goldbattle, 2020-09-11:
Yes, we have used it fine with a single camera and IMU without issue. You should just be able to do multi-cam calibration, then the imu-camera. The only difference is you only specify one camera topic and model.

I recommend you open an issue on Kalibr's github if you keep running into issues, but I recommend you check out their wikis:
https://github.com/ethz-asl/kalibr/wiki/multiple-camera-calibration

SubMishMar, 2020-09-11:
Thanks for your response.

Yes I ran Kalibr just with one camera (one topic and one model...) and the snippets above show what I got. Right now I am using the results from the first snippet, ie. the values it initialized with. I am not sure if it is the way to go. I can always use the ros camera calibrator. 

It might be a visualization issue, but to me it seems more like kalibr wants multiple camera (Ref to MulticamGraph.py).  I am asking my questions here because the Kalibr repo seems to be pretty dormant when it comes to answering queries and since you people have put up such a nice website illustrating Kalibr's use, I thought this is the right place to ask questions.

I tried running the code by using flags like --dont-show-report and yet, I get the same errors (warning).

Another question, and this is related to the sensor calibration tutorial. You say that the shutter_time should be low to reduce motion blur? Currently I am using a Chameleon3 PGrey camera with auto_exposure/auto_shutter/auto_gain etc disabled. The shutter_time is 0.03 ms and yet I see significant motion blur. (My camera data is coming at 30 Hz). Do you have a ball park number which works well? Should reducing the camera frequency from 30 Hz to 20 Hz reduce motion blur?

Thanks for your time.

SubMishMar, 2020-09-11:
So, I recalibrated the system with low shutter time (0.01 s) and the results I get are decent (not sure if acceptable). Do you know what is an acceptable value of camera re-projection error? For my setup, it is around 1.6 for a camera of resolution 1288 x 964. 

Thanks.

goldbattle, 2020-09-12:
I dug through some of my old files and this is one of the launch file examples I have for pointgrey.
https://gist.github.com/goldbattle/3cc02e2e7477b16070fa7c1794feff13

We normally use 640x480, as processing at higher resolution becomes very limited due to the speed of the frontend tracking in OpenCV. OpenVINS handles async camera-imu sensors, with arbitrary timeoffsets. You just need to enable this online estimation in the launch file (look at the example ETH launch file for examples) or if you know its offset you can just input it in the launch file and not estimate it online (not recommended).

SubMishMar, 2020-09-12:
Thanks @goldbattle ! Appreciate it!

closed 2020-09-15 by goldbattle