rpng/open_vins #99: OpenCV Version

mhuzai, 2020-09-11:
The [installation guide](https://docs.openvins.com/gs-installing.html) states OpenCV 3.4.6 as a dependency. I was wondering why this particular version of OpenCV is required, and if there are any plans to move to a newer version of OpenCV.

Thank you!

goldbattle, 2020-09-12:
It was just what was developed with. I believe I tried building it with OpenCV 4 and everything worked out ok if I remember correctly. In terms of support, having a specific supported version so I don't have to deal with support issues about OpenCV installs is mainly why we just chose what we used at the time.

mhuzai, 2020-09-12:
Gotcha. Good to know that we can upgrade if the need arises. If we do end up moving to CV4, would an upgrade PR be of use to you?

goldbattle, 2020-09-15:
Feel free to open a PR if you feel the need, I am not sure if we would merge it but it might be helpful for others.

closed 2020-09-15 by goldbattle